firmata/firmata.js #85: Board.prototype.i2cConfig is redundant

monteslu, 2015-02-12:
It does the same thing as Board.prototype.sendI2CConfig

Any reason it should exist?

rwaldron, 2015-02-12:
Yes, because I deprecated all of the `sendI2C...` APIs, so for consistency, `sendI2CConfig` => `i2cConfig`

rwaldron, 2015-02-12:
Also: eventually, there will be console warnings for using those APIs

closed 2015-02-12 by rwaldron

achingbrain, 2015-02-17:
Presumably the one-wire methods should be renamed too?

`sendOneWireConfig(pin, enableParasiticPower)` => `oneWireConfig(pin, enableParasiticPower)`

rwaldron, 2015-02-17:
@achingbrain I was thinking about this and have a "note to self" to ping you about it. Some of the things in the API should be cleaned up as well, eg. the boolean trap arguments should be cleaned up

With the example you gave, I want to change that to: 

```js
Board.prototype.oneWireConfig = function(opts) {
  ...
};

Board.prototype.sendOneWireConfig = function(pin, enableParasiticPower) {
  console.log("Deprecated...");
  
  return this.oneWireConfig({
    pin: pin, 
    enableParasiticPower: enableParasiticPower
  });
};