firmata/firmata.js #87: How to reconnect after USB cable is unplugged?

fsamir, 2015-03-18:
Hi everyone,

is there an example on how to restart Board, once the USB cable is unplugged and plugged again?

Thanks

soundanalogous, 2015-03-18:
The board is restarted automatically once you the serial connection is established. State is not saved on the board however so each time you unplug it all state will be lost.

fsamir, 2015-03-21:
Where the reconnects are done? I can't see it firmata.js nor serialPort and when I simply reconnect the cable with the blink example, it never reconnects.
I have only been able to reconnect by calling `serialPort.close()` upon an error and `serialPort.open()` on an reconnect loop.

soundanalogous, 2015-03-21:
I thought you were talking about restarting the physical board from software. When the board is reconnected you will have lost state since it is not stored on the board. Therefore there's no advantage in an auto reconnect with USB since you'd be starting over each time you reconnect.