firmata/firmata.js #88: How do you close the connection?

digihaven, 2015-03-23:
How do you close the connection to allow Nodejs to have a clean shut down? There seems to be a missing .close() function that should be implemented for this firmata module.

The only way I could figure out, is by creating the socket my self. But there are (setTimeouts??) that cause crashing:
```
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: Error Bad file descriptor calling write(...)
    at Error (native)
```

Any better suggestions?

vicatcu, 2016-04-11:
I have more or less the same question, if I create the Board object like this:

``` javascript
var myBoard = null;
function initializeMyBoard(){
  Board.requestPort( function(err, port){
      if(err){
        console.log(err);
        return;
      }

      console.log("Acceptable Port Found");
      myPort = port.comName;
      var board = new Board(myPort);
      board.on("ready", function(){
        console.log("Board Ready!");
        myBoard = board;
      });
  });
}

// shouldn't I be able to call some function on myBoard
// down here to close the connection and gracefully
// shutdown the serial port?

// so that I can do some other stuff at this point 
// without being connected to the Serial Port and then 
// call initializeMyBoard() when I feel like it?
```

dceejay, 2016-07-12:
I think you can call board.sp.close()  to close the underlying serialport