firmata/firmata.js #89: Can't read analog value (node-webkit 0.12.0 & serialport v1.6.3)

EdisonDelgado, 2015-04-17:
Hello, I try to get the last value from the board object  but always is = 0
```
board.pins[board.analogPins[0]].value
```
```
board.analogRead(pin, function(value){
     console.log(value);
})
```
last code throw "Cannot set property 'report' of undefined"

Thank in advance.

rwaldron, 2015-04-17:
Can you paste in your complete code? Thanks!

EdisonDelgado, 2015-04-17:
This function connect my Arduino board.
```
function connect(){
	if(board){
		board.sp.close();
		board = undefined;
	}
		
	path = combobox.val();
	console.log("connecting to "+path);

	board = new firmata.Board( path ,function(err){
		if(!err){
		console.log("connected to "+path);
		board.sp.on("disconnect" ,handleDisconnection);
		board.sp.on("error", function(err){ handleError(err)});

		

		Testhandler = setInterval(runTest, 500);
		}
		else{			
			return showMessage(err.message);
		}		
	});
}
```
this function is triggered every 500ms and try to get last analogs values.
```
function runTest(){

	if(board){
		temperature.text(board.pins[board.analogPins[0]].value.toString());
		light.text(board.pins[board.analogPins[1]].value.toString());
		audio.text(board.pins[board.analogPins[2]].value.toString());
		humidity.text(board.pins[board.analogPins[3]].value.toString());
		infrared.text(board.pins[board.analogPins[4]].value.toString());
		slider.text(board.pins[board.analogPins[5]].value.toString());

		console.log("running");
	}
}
```

When i write digital pins works fine
```
$('#buzzer').on("slide", function(pwm){
		board.pinMode(6, board.MODES.PWM);
		board.analogWrite(6, pwm.value);
});

$('#whiteLed').change(function(){
	var val =  $(this).prop('checked');
	if(board){
		board.digitalWrite(13, val);
	}
});
```
[image]

EdisonDelgado, 2015-04-17:
In older versions (serialport & node-wekit)  this code works fine.

Before reading the pin Analog I forgot configure.
```
board.pinMode(pin, board.MODES.ANALOG);
``` 
Now is working!

Thanks!

closed 2015-04-17 by EdisonDelgado