firmata/firmata.js #94: Breaking change from samplingInterval

mgcrea, 2015-06-15:
After updating to the latest releases, my arduino-like boards stopped working. Did bisect the regression to this commit: 49aab8d4975434f6ebd29e747777407d16ddd07e.

If I configure a board to spam a simple json-like "hello world" every 3 secs, I only get the first one.

What is `samplingInterval` doing behind the scene that could explain this?

rwaldron, 2015-06-15:
I'll take a look at this today

rwaldron, 2015-06-15:
> If I configure a board to spam a simple json-like "hello world" every 3 secs, I only get the first one.

Can you explain this a bit more? Any code you can share will be helpful.

soundanalogous, 2015-06-15:
In [this line](https://github.com/jgautier/firmata/blob/49aab8d4975434f6ebd29e747777407d16ddd07e/lib/firmata.js#L1235), specifically this code: `((safeint >> 8) & 0xFF)` it appears that the sampling interval will be set to `0` if any value less than 255 is passed. I'm not sure why the bit shifting is performed here. That's likely the cause.

soundanalogous, 2015-06-15:
nm... I overlooked the 3rd element that sets the proper value. However safeint should be shifted 7 bits rather than 8 bits and the first byte should be masked against 0x7F rather than 0xFF:

`this.transport.write(new Buffer([START_SYSEX, SAMPLING_INTERVAL, (safeint & 0x7F), ((safeint >> 7) & 0x7F), END_SYSEX]));`

rwaldron, 2015-06-16:
@soundanalogous I discovered an additional issue that I missed when https://github.com/jgautier/firmata/commit/a448356bbc060b26550f50f60f8d63d94d5698bb landed: `setSamplingInterval` was previously only called when user code explicitly called it, but that patch resulted in it _always_ being called because it `settings.samplingInterval` always had a default. I've fixed this to check for explicitly defined `options.samplingInterval` instead. I've also fixed the issue safeint shifting.

closed 2015-06-16 by rwaldron