firmata/firmata.js #95: Uart/serial support

rwaldron, 2015-06-24:
Depends on https://github.com/firmata/arduino/pull/205

Here's my rough game plan:

- Report to @soundanalogous throughout the process (Basically, insert this item between all of the following bullets)
- I'm working on a survey of all platforms that have uart/serial support. 
- From there I'll do a draft API and seek team feedback.
- Attempt to implement API in all surveyed platforms. 
- Extensive component testing against those platforms to refine API and hopefully iron out major issues
- Team component testing (people should make lists of hardware they own and post that list here as a check list)

rwaldron, 2015-06-24:
- [ ] ITEAD GPS shield 1.1
- [ ] Generic GPS breakout
- [ ] Adafruit GPS shield 
- [ ] Adafruit GPS Breakout v3
- [ ] SSC-32 servo controller
- [ ] SSC-32U servo controller
- [ ] Pololu 8 channel servo controller

soundanalogous, 2015-06-24:
- [ ] ID-12 RFID reader

soundanalogous, 2015-06-24:
Potential issues to look for when testing these components with an Arduino:
- Overrun serial write or read buffer. If the device you're trying to support requires reading or writing more than 64 bytes at a time, bytes may be lost. Try to break the read or write into smaller chunks if possible. If this is not possible for some devices and you notice issues, please report this.
- Poor performance of non-serial devices attached to the board. Reading and writing large serial buffers can potentially degrade responsiveness. If analog I/O becomes twitchy or you notice other non-normal behavior of I2C devices, etc while writing or reading large serial buffers, please report this. There is a currently unused value in the protocol that can be used to limit the number of bytes read per iteration of the main loop (running on the Arduino) that is reserved just for this purpose.

Some tips:
- Pay attention to the pin limitations on Mega, Leonardo and Micro boards when using SoftwareSerial: http://www.arduino.cc/en/Reference/SoftwareSerial.
- Favor HW serial over SW serial for boards that support it (Leonardo, Mega, Due, etc).

rwaldron, 2015-08-05:
> Poor performance of non-serial devices attached to the board. 

Working on this next

rwaldron, 2015-08-06:
@soundanalogous 

> Poor performance of non-serial devices attached to the board. Reading and writing large serial buffers can potentially degrade responsiveness. If analog I/O becomes twitchy or you notice other [...]

I think this is going to be necessary. I put together a performance [measurement suite](https://github.com/rwaldron/firmata/tree/serial-performance) and made a couple notable observations: 

- Serial write will slow down I2C write enough to be visibly noticeable. I was "flashing" a BlinkM in 20ms on/off intervals while writing 4 bytes to a software serial port that had nothing connected to it; the result was consistent in that the serial writes made the I2C writes appear inconsistent. 
- Serial read can potentially bog the program down. When I attached a GPS breakout, there is no difference in the digital, analog or I2C reads before the device gets a satellite fix. Once it has the fix, there's an immediate observable difference in the read timing (illustrated in the data I've pushed up)
- When I was trying to have two software serial ports in service, I was no longer able to read from the first one configured.

https://github.com/rwaldron/firmata/tree/serial-performance

soundanalogous, 2015-08-06:
> When I was trying to have two software serial ports in service, I was no longer able to read from the first one configured.

You can only read from one SW serial device at a time. This is a limitation of the Arduino SoftwareSerial library. You need to call the `listen` method to switch between configured SW serial ports.

rwaldron, 2015-08-07:
> You can only read from one SW serial device at a time. This is a limitation of the Arduino SoftwareSerial library

Ah, got it. This probably best addressed with documentation. 

I'll make some changes to the component perf program and report back.

soundanalogous, 2015-08-09:
It appears the issue with serialWrite in the measurement test actually has something to do with reading pin 9. Comment out line 189 in components.js and uncomment line 196 but change to SW_SERIAL0 and you'll see a huge improvement. I suspect reading pin 9 may affect software serial (on an Uno at least) because the timer used for SW serial on pin 10 is likely also related to pin 9, even if pin 9 is just doing digital I/O rather than PWM. Interesting indeed...

This measurement suite is great BTW! Will be interesting to use this to compare performance across boards as well (Due vs Uno for example).

soundanalogous, 2015-08-09:
I'll implement bytesToRead on the Arduino side (currently it's ignored) to see if that helps with performance. Passing a value of 0 will read all available bytes in the serial buffer on each iteration of the main loop whereas passing a value > 0 will read only that number of bytes per iteration. Passing too low of a value for a high baud interface could result in lost data so this will probably be an advanced feature as it will require tuning per attached serial device. The Adafruit GPS is a great test though since it dumps so much data.

soundanalogous, 2015-08-09:
Strange... it's actually writing to pin 8 that causes the issue (regardless of which pin is read). Not sure why. I changed the 2nd digital write to pin 4 and read to pin 6 and that works fine. I see a 1 character error in the serial write output (I attached a serial USB console cable to pin 11) when the GPS dumps it's data but otherwise write looks good even with i2c, analog read and 2 digital write / read.

soundanalogous, 2015-08-09:
It appears what is happening is SoftareSerial uses pin change interrupts. Pin change interrupts are handled per port. On an Arduino Uno (or other ATMega328p board) pins 8 - 13 are on the same port (and pins 0 - 7 are on another port). Using SoftwareSerial on any pin in that port will enable the pin change interrupt on that port. The interrupt is triggered any time a digital pin is toggled on that port, thus toggling pin 8 would trigger the interrupt constantly and that will mess up read or write for everything else.

So it looks like when using SoftwareSerial, the user cannot toggle digital pins on the same port as the software serial device. This will be difficult to enforce for some devices because the microcontroller port does not always map 1-to-1 to Firmata ports.  For ATMega328p devices the ports actually do align but for most other devices they do not so there would be not way to programmatically enforce a rule about not using digital pins on the same port as a SW serial pin.

soundanalogous, 2015-08-09:
Testing with an Arduino Mega, HW serial performance looks much better. Still seeing some loss in serial write output, but lapse time between analog and digital reads during and just after serial reads looks much better than with SW serial (as expected).

rwaldron, 2015-08-10:
> Passing a value of 0 will read all available bytes in the serial buffer on each iteration of the main loop whereas passing a value > 0 will read only that number of bytes per iteration

Sounds good to me. 

> This measurement suite is great BTW! Will be interesting to use this to compare performance across boards as well (Due vs Uno for example).

Thanks! I have some updates to push based on all of your feedback here.

Can you push up the latest to https://github.com/firmata/arduino/pull/205/files

soundanalogous, 2015-08-10:
I haven't made any changes on the Arduino side yet regarding bytesToRead.

If you have a Leonardo, you should give that a try with the measurement suite (wiring to RX1 and TX1 - HW_SERIAL1). It shows even better performance than an Arduino Mega using HW pins. One change you'll need to make is you can't use D2 or D3 on a Leonardo while using I2C devices (because D2 and D3 are I2C pins but unlabeled as such).

rwaldron, 2015-08-10:
> I haven't made any changes on the Arduino side yet regarding bytesToRead.

Got it, I wasn't sure if you had implemented that yet. 

> If you have a Leonardo, you should give that a try with the measurement suite (wiring to RX1 and TX1 - HW_SERIAL1). It shows even better performance than an Arduino Mega using HW pins.

Cool, I'll give it a try.

> One change you'll need to make is you can't use D2 or D3 on a Leonardo while using I2C devices (because D2 and D3 are I2C pins but unlabeled as such).

How nice of them :(

soundanalogous, 2015-08-16:
I've implemented bytesToRead on the Arduino side (sync the serial branch). Appears to make minor improvements when setting a low value like 1 for bytesToRead when using SW serial RW in components.js.

rwaldron, 2015-08-16:
Awesome, I will continue testing out different components, but I think this is about ready to land :)

soundanalogous, 2015-08-16:
If you have any serial devices that use a baud rate of 57600 or higher that would be ideal for testing. At higher baud rates it's more likely that data could be lost and where the bytesToRead value (if used) will require finer tuning.

rwaldron, 2015-08-17:
Ok, I'll take a look at see what I have

soundanalogous, 2015-08-24:
I made some [minor updates to firmata.js](https://github.com/soundanalogous/firmata/commit/78c24e84d40d4e013ad6f15b123a614ab51f5ea8), including a Serial pin mode.

rwaldron, 2015-10-01:
> I've implemented bytesToRead on the Arduino side (sync the serial branch). 

There appear to be two:

- `serial`
- `serial-options` (has all of `serial`, with newer commits)

Questions: 

- Should we work from `serial-options` now? 
- Have you updated firmata.js w/r to `bytesToRead` changes? If not, would you like me to take it on?

soundanalogous, 2015-10-01:
I have made the changes locally but have not pushed them yet (I've been on vacation for the past 2.5 weeks and just got back this week).

While I've moved `bytesToRead` (renamed to `maxBytesToRead` which is more accurate) to the `serialRead` method I'm not sure if this is necessarily the best chance since it's meaning is different than bytesToRead in I2C. With I2C, `bytesToRead` should always return the specified number of bytes unless there is an error, with Serial however it will return a maximum of the number of bytes specified, but depending on how many bytes remain in the buffer, it may return fewer bytes. What are your thoughts here? I do like having the ability to change the value easily via serialRead rather than locking it into the constructor.

Regarding serial-options. It turned out that serial options (data bits, parity, stop bits) was more difficult to implement on the Arduino side than I had initially anticipated. The issue is that Arduino does not maintain a consistent signature for `Serial.begin(baud, options)` across all architectures. For AVR boards, options is an int. For SAM boards, options is one of two possible Enums. The code required to resolve this is [ugly as hell](https://github.com/firmata/arduino/blob/serial-options/utility/serialUtils.h#L124-L349) and fragile (in the case of new architectures or compatible 3rd party boards using yet another signature). I'm not sure yet how to proceed. Having these options would be helpful although the need for them is really more of an edge case. My biggest concern is how fragile (and messy) the implementation is.

soundanalogous, 2015-10-01:
This is actually the serial options code that I'm worried about, not what I referenced in the previous comment (that was ugly but stable). This however could break with a new board, etc: https://github.com/firmata/arduino/blob/serial-options/examples/StandardFirmata/StandardFirmata.ino#L748-L767

rwaldron, 2015-10-01:
> While I've moved bytesToRead (renamed to maxBytesToRead which is more accurate) to the serialRead method I'm not sure if this is necessarily the best chance since it's meaning is different than [...]

This is a super valuable distinction. I've been thinking of this change as being somehow analogous to the i2c bytesToRead—I'm glad you called this out. 

> What are your thoughts here? I do like having the ability to change the value easily via serialRead rather than locking it into the constructor.

I still like it, for the same reason you've said here. I think the distinction above is simply a matter of documentation. 

> https://github.com/firmata/arduino/blob/serial-options/utility/serialUtils.h#L124-L349

Holy shit.

>  I'm not sure yet how to proceed. Having these options would be helpful although the need for them is really more of an edge case.

If you asked me to "vote", I'd say punt until someone comes along to complain about it. In all the testing of various servo controllers, gps, rf readers, rc receivers (and so on) that I tested with an arduino, I encountered no resistance.

soundanalogous, 2015-10-02:
I'll make the changes this weekend. Going to remove serial options for now since I can add it later if necessary in a backwards compatible way.

soundanalogous, 2015-10-03:
Updated both [node-firmata (my branch)](https://github.com/soundanalogous/firmata/commit/b1666ca7b224c138d499df688cee187f6817b25e) and [Firmata](https://github.com/firmata/arduino/commit/0e3a973877ddd44392ba24196df9d10d1434ac89). This should be final as far as the Firmata serial protocol goes.

We should decide on which examples to include.

soundanalogous, 2015-10-03:
@rwaldron maybe a good next step is for me to submit a PR to add my serial branch to a serial branch in this repo?

soundanalogous, 2015-10-07:
Serial support is now in Firmata master. We can add serial to node-firmata now.

rwaldron, 2015-10-27:
:clap: @soundanalogous 

Just released 0.8.0 with serial support

closed 2015-10-27 by rwaldron