firmata/firmata.js #100: Can has tags?

aredridel, 2015-10-18:
It would be great to have tags in the github repo for comparing to released versions. The published git hashes are:

```
0.3.4 70155fa9abf2bc07520f1467c71c12dde8da6149
0.3.5 c8b9edd837d282ff6be41c6c761e1555d38afa43
0.3.6 f30e9c8ce62379d7e102ba248f2aa47cf3aad105
0.3.7 36d02dfff262d6b5744062aab26d5aab711d98c5
0.3.8 0fe062810b75650cb92873d4a113ed74d7454863
0.4.0 a40bf8e510234afc55a869d6ed441744bfedf38e
0.4.1 cf1d4de658f273a062455bb6479d33c9197e5ef2
0.5.0 d862ee5deb32d65d0099dc0fd029020c8d8fd84b
0.5.1 f7b024b9b6086dcca12e23ab40536a8c075b8a99
0.5.2 56ffd19f9fe63010b1d1b62fbfa175b824be778e
0.5.3 c7079de46b11b0d89ae4bffa7af3fdc00cc6ef95
0.5.4 ff93d4e5dcebcd61acfa648db54cac2b1bfda982
0.5.5 3e093e43c9d9847281854e7ffc4313764a3b9486
0.6.0 41f28f94c299d5a02b74b2579645766a31b561ca
0.7.0-pre-DO-NOT-USE da01d6f4f4c0ddb12c6cd92eac988d395b16c68f
0.7.1-pre-DO-NOT-USE 8c99c24d8661a00430a276f2a6f936724b3b66e8
```

rwaldron, 2015-10-20:
https://github.com/jgautier/firmata/releases

closed 2015-10-20 by rwaldron

aredridel, 2015-10-21:
\o/