firmata/firmata.js #102: Can't run examples with port.comName

adrianardu, 2015-12-03:
Hi if I try to run the blink example with node blink.js, I get the following error from my arduino.

/usr/lib/node_modules/firmata/examples/blink.js:6
    var board = new Board(port.comName);
                                                ^
TypeError: Cannot read property 'comName' of null
    at /usr/lib/node_modules/firmata/examples/blink.js:6:31
    at /usr/lib/node_modules/firmata/lib/firmata.js:1698:7
    at /usr/lib/node_modules/serialport/serialport.js:451:18
    at Object.oncomplete (fs.js:107:15)

Thanks for you help.

rwaldron, 2015-12-03:
What version of node? What OS?

rwaldron, 2015-12-03:
I've added error handling to all the examples using `Board.requestPort`

adrianardu, 2015-12-04:
it is node v0.10.33. It is installed on a arduino yun.

adrianardu, 2015-12-04:
ok if I put directly after the Board.requestPort a console.log I get:

[Error: No Acceptable Port Found]

It is plugged with USB on COM4 Port. Shouldn't it be found automatically?

thanks
-Adrian

rwaldron, 2015-12-04:
These pieces of information don't quite fit together: 

> it is node v0.10.33. It is installed on a arduino yun.

And...

> It is plugged with USB on COM4 Port. Shouldn't it be found automatically?

There is no such port on an Arduino Yun (COM* is windows-specific.) 

> [Error: No Acceptable Port Found]

If there was a port that Firmata.js could connect to, Serialport would provide it. Try this: 

```js
var Serialport = require("serialport");

Serialport.list(function(error, ports) {
  if (error) {
    console.log(error);
    return;
  }

  console.log(ports);
});
```

adrianardu, 2015-12-07:
-Ok, I forget. I installed node in ubuntu 14.04, because npm doesn't work on arduino.

I tried you code and it returns me only: "[]"

rwaldron, 2015-12-07:
That means that the `Serialport` module itself cannot find any ports. Please open an issue here: https://github.com/voodootikigod/node-serialport/issues

closed 2015-12-07 by rwaldron