firmata/firmata.js #103: Dipping toes into serial but am stuck (question)

dtex, 2015-12-21:
I’m trying to run this firmata example as is https://github.com/jgautier/firmata/blob/master/examples/sw-serial-read-gps.js but am not seeing any data. 

I’m using standardFirmataPlus 2.5 on the board and firmata 0.8.0 on the host server. I was able to confirm my GPS is working by hooking it up to pins 0 and 1 and listening with serial monitor.

First I just want to reproduce what y’all did. Can I ask which GPS module you tested with? Perhaps mine requires some initialization stuff that y’alls did not?

dtex, 2015-12-21:
Nevermind. I had my wired crossed (literally)

closed 2015-12-21 by dtex

soundanalogous, 2015-12-21:
One thing that would be a nice add for that example would be to parse out the results rather than simply printing the bytes as they arrive.

dtex, 2015-12-21:
Maybe basic lat and long? On the other hand, I think anyone digging in at this level would be perfectly happy to just see the bytes roll on by... I know I was :-)

So here's the thing threw me... When wiring up to pins 0 and 1 on my uno, tx on the GPS was connected to tx on the Uno and rx to rx. That configuration let me see the data in Serial Monitor. 

For sotwareSerial I had to have rx going to tx and tx to rx. 

Why the difference?

soundanalogous, 2015-12-21:
It doesn't work that way for me. I wire TX on the GPS to RX on the Arduino for either SW serial or HW serial.

soundanalogous, 2015-12-21:
Also when you say you wired the GPS to pins 0 and 1 on your Uno that you're not running Firmata. If you were you'd be seeing the Firmata output in Serial Monitor, not the GPS output.

dtex, 2015-12-21:
That's right, I switched to an "empty" sketch when I first started testing the GPS and listening on serial monitor, but was running standardFirmataPlus when using softwareSerial. 

I'm sure that earlier bypass was the source of my confusion.

Thanks so much for this. I'm working on a GPS class for Johnny-Five and support for Dynamixel servos next.

dtex, 2015-12-22:
So what's the over/under on serial making it into standardFirmata?

rwaldron, 2015-12-22:
It's in StandardFirmataPlus, which is now the instructed version to use: https://github.com/rwaldron/johnny-five#setup-and-assemble-arduino