firmata/firmata.js #104: Changing baud rate in softwareSerial

dtex, 2016-01-02:
### My Scenario

Using an Arduino Uno and softwareSerial I'm trying to connect to a device that runs at 115200 baud by default. I know the Uno & softwareSerial cannot handle that baud rate reliably so the first thing I do is send a short command to change the device to 9600 baud, then close my connection and reopen at 9600. 

### Sample Arduino code

This sample works fine:

````C
#include <SoftwareSerial.h>
SoftwareSerial mySerial(11, 10); // RX, TX

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(57600);
  
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

mySerial.begin(115200);
  
  // Send 32 byte command to change message type and baud rate
  byte changeOpMode[] = {0xA0, 0xA2, 0x00, 0x18, 0x81, 0x02, 0x01, 0x01, 0x00, 0x01, 0x01, 0x01, 0x05, 0x01, 0x01, 0x01, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01, 0x25, 0x80, 0x01, 0x3A, 0xB0, 0xB3 };
  mySerial.write(changeOpMode, sizeof(changeOpMode));
  mySerial.end();
  
  // Reopen connection at new baud rate
  mySerial.begin(9600);
}

void loop() { // run over and over
  if (mySerial.available()) {
    Serial.write(mySerial.read());
  }
  if (Serial.available()) {
    mySerial.write(Serial.read());
  }
}
````

dtex, 2016-01-02:
Before you dig in, I need to write a simple test using just firmata JS. My current tests are using J5 which obfuscates things a bit.

soundanalogous, 2016-01-02:
The root of this issue is on the Arduino (c/c++) side, not the JavaScript side. I can see now based on your Arduino example above what is different from my implementation. I'm deallocating the pointer to the SoftwareSerial object when calling `end()`, then if the user tries to config again, a new SoftwareSerial instance is assigned to the pointer that must be breaking something (and could also lead to a heap and stack collision if done multiple times). Whereas in your example above you simply call `end`, then `begin` with a new baud rate.

I want to keep the ability to deallocate memory in case a serial port is no longer needed after a certain point in an application. What I could do is add a new message that enables the user to update the baud rate. This would essentially call `end` then `begin` with the new baud as in your example. The `close` would deallocate.

dtex, 2016-01-02:
That would be perfect for me, hopefully wouldn't bloat standardFirmataPlus much and would open up more serial devices for users. I'd like to take a stab at adding this but I suspect you've already written 90% of it in your head.

dtex, 2016-01-02:
Gonna open this over at firmata/arduino instead

closed 2016-01-02 by dtex