firmata/firmata.js #106: Problems with pins[pin].value

jguille2, 2016-01-09:
Hello,
**I can't get the real pins value in firmata 0.8.0**
My test...

---
With **firmata 0.3.3** I use **pins[pin].value** and I get the **right value**.

    var firmata = require('firmata');
    board = new firmata.Board(portName, function(err) {console.log(err);});
    console.log(board.pins[pinNumber].value);

---
With **firmata 0.8.0 always report me a zero value**.
In this version I can  do

      board = new firmata.Board(portName, function(err) {console.log(err);});
or

      board = new firmata(portName);

but in both cases I get the same: *value = 0* !!!

rwaldron, 2016-01-09:
Thanks for the report.

The example given:

```js   
var firmata = require('firmata');
board = new firmata.Board(portName, function(err) {console.log(err);});
console.log(board.pins[pinNumber].value);
``` 
... is not a valid program—here's why:

- `portName` is `undefined` here, so this is never going to connect to anything:
  ```js
  board = new firmata.Board(portName, function(err) {console.log(err);});
  ```
- the `pins` property can't possibly have any items in it because those items are created after the Board has connected, which it hasn't. Even if the `portName` wasn't `undefined`, the connection operation is asynchronous, and isn't complete until the callback above is called. The line is always executed before the connection is complete:
  ```js
  console.log(board.pins[pinNumber].value);
  ```
- `pinNumber` is also `undefined`

The tests for pin value have not changed between those versions cited, nor is there history of failure so I'm going to close his for now. If you can provide a valid test case that demonstrates a bug, we can reopen and revisit.

jguille2, 2016-01-09:
Ups!
I writed *portName* and *pinNumber* to show a generic code. Obviously these two vars have a value in my code.

As I said, this code is running ok with Firmata 0.3.3. With 0.8.0 (the same code, only changing the node_modules folder) is running (without errors) but value always is zero.

jguille2, 2016-01-10:
Thanks Rick for your work!

I don't know if there is an issue or it's the normal operation.

I have to do more tests... because I can not reproduce now with a fresh install (I had a 0.3.3 installed for nw.js and maybe it was hacked).

Only a question. For digital pins in input mode, is pins[ ].value auto updated?

rwaldron, 2016-01-11:
> `is pins[ ].value auto updated`

Yes, https://github.com/jgautier/firmata/blob/master/lib/firmata.js#L151-L158

jguille2, 2016-01-13:
Thank's Rick,

But I have a question. Your link shows that pins[ ].value is update with *DIGITAL_MESSAGE*. I think that this update will be done with digitalRead calls.

**Without digitalRead call, pins[ ].value isn't update, isn't it?**

I tested and I've seen that (without digitalRead call) pins[ ].value don't show the real value of Input pins. Input pins show the 'state' of the pullup internal resistor, because firmata.js change its value in digitalWrite function. For output pins it's ok.

I've tested more and I see with Firmata 0.3.3 there was a diferent operation mode and pins[ ].value shows always the real value of the pin.

Maybe it is not an issue... only an operation mode change. But it's an important change... and it must be clear.

Thank for your work an patience :)

rwaldron, 2016-01-13:
> Without digitalRead call, pins[ ].value isn't update, isn't it?

Yes, it is: https://github.com/jgautier/firmata/blob/master/lib/firmata.js#L777

> Maybe it is not an issue... only an operation mode change. But it's an important change... and it must be clear.

But there hasn't been any change to these operations between 0.3.3 and 0.8.0, take a look: 

- 0.3.3
  - [DIGITAL_MESSAGE](https://github.com/jgautier/firmata/blob/4e9568be368984907919d255fc9b1d11580de776/lib/firmata.js#L129-L137)
  - [digitalWrite](https://github.com/jgautier/firmata/blob/4e9568be368984907919d255fc9b1d11580de776/lib/firmata.js#L613)

- 0.8.0
  - [DIGITAL_MESSAGE](https://github.com/jgautier/firmata/blob/3c164bdcf2401ac2fcfb22bc9c57fcbce62e5798/lib/firmata.js#L151-L159)
  - [digitalWrite](https://github.com/jgautier/firmata/blob/3c164bdcf2401ac2fcfb22bc9c57fcbce62e5798/lib/firmata.js#L777)

jguille2, 2016-01-14:
Hi Rick and I'm sorry for making you waste time ... but I think I have not explained well enough.

What I see:

- [Your link](https://github.com/jgautier/firmata/blob/master/lib/firmata.js#L777) talk about updating *value* in *digitalWrite* function. This is ok. Right for *OUTPUT* pins, but for *INPUT* pins this shows the state (if the *pull up* internal resistor is tured on/off), not the value.
- I think that changes between 0.3.3 i 0.3.4 affects these operations. In [this issue](https://github.com/jgautier/firmata/pull/58) is representative. Maybe [this commit](https://github.com/jgautier/firmata/commit/17dba504fcb05efb2fa4d197b376802fb4bebe7d) shows (maybe is changed) the operation mode. Then, without digitalRead calling, pins[ ].value wasn't update.

rwaldron, 2016-01-14:
Ok, I see the problem now—thank you for your patience. Yes, this breakage was overlooked

rwaldron, 2016-01-14:
Ok, I see the problem now—thank you for your patience. Yes, this breakage was overlooked. I will work on implementing a way to optionally report all values on initialization.

rwaldron, 2016-01-14:
I'm not sure what's going on here. Using this basic hardware: 

[image]

I checkout to 0.3.3 and run the following program: 

```js
var Board = require("../").Board;

var board = new Board("/dev/cu.usbmodem1411", function() {
  console.log(board.pins[8].value === 1 ? "PASS" : "FAIL");
});
```

You're asserting that the value should be 1? Using 0.3.3, that never passes for me:

- `board.pins` doesn't have any entries yet: https://github.com/jgautier/firmata/blob/4e9568be368984907919d255fc9b1d11580de776/lib/firmata.js#L129-L137 
- ... because the `queryCapabilities` operation hasn't completed: 
  - https://github.com/jgautier/firmata/blob/4e9568be368984907919d255fc9b1d11580de776/lib/firmata.js#L490-L510
  - https://github.com/jgautier/firmata/blob/4e9568be368984907919d255fc9b1d11580de776/lib/firmata.js#L637-L640
  - https://github.com/jgautier/firmata/blob/4e9568be368984907919d255fc9b1d11580de776/lib/firmata.js#L184-L203

jguille2, 2016-01-16:
Hello Rick,
Oh... Maybe it isn't an issue... only a question... but I try to add info.

**There two different things here. To make an asynchronous script and the changes between 0.3.3 and 0.3.4.**

-Your code without *digitalRead* shows always a *FAIL* value. It doesn't show the real value of the pin. It is for all versions (0.3.3 and newer versions).

-Your code with *digitalRead* is ok (showing real value of the pin) in all versions too.

-But if I call directly (without *digitalRead*) into an asynchronous script, things change. I made an example making code into a webServer...
    
    var firmata = require("firmata");
    var webServer = require("http");
    var board = new firmata.Board("/dev/ttyACM0",function(){
      console.log("board ready");
      serverInit();
    });
    function serverInit(){
      webServer.createServer(function (request, response) {
        response.writeHead(200, {'Content-Type': 'text/html','Access-Control-Allow-Origin': '*'});
        response.write("Writing.. 8 pin value: ");
        board.pinMode(8,board.MODES.INPUT);
        response.write(board.pins[8].value.toString());
        console.log(board.pins[8].value);
        response.end();
      }).listen(8111);
    }

-If I write in my browser *http://localhost:8111* I'll get the pins value (in my browser and in the console). But here is the change!! This code is OK for version 0.3.3 (and older). For newer versions (0.3.4+)  it's not running (always zero values). And then, for newer version, I must change my code and use *digitalRead*.

I hope this can clarify something. At least I've learned more clearly this operation. Maybe we can add this info to docs... It say *value: Number, // Current value of the pin...* and it's important the *digitalRead* required for input pins.

soundanalogous, 2016-01-16:
@jguille2 you are not getting a pin value because pin reporting is not enabled by default since v0.3.4. The Firmata firmware running on the arduino board will only report pin values if pin reporting is enabled for this pin. The call to enable pin reporting is:

```javascript
// enable digital pin values to be sent for pin 8
board.reportDigitalPin(8, 1);
```
However like all things in Firmata, this call is asynchronous so you cannot use `board.pins[8].value` to read the value until the message has been received by the firmware and the value of the pin has changed. Since you will not know when this happens, it is best to use `digitalRead`. The `digitalRead` method first enables pin reporting for that pin then sets a listener to be called whenever the digital input value changes.

jguille2, 2016-01-16:
Thank's Jeff!!

Your post answers my question :joy: !!
I needed to confirm the change in these operation after 0.3.3 version. I had not seen *reportDigitalPin* function. This is the solution I needed.

I understand (more or less) the asynchronous behavior... and that's why i wrote the example code of the webserver.

I think this issue is closed. Perhaps we should add some notes to documents showing this operation: the requirement to use *digitalRead* or *reportDigitalPin* to update *pins[ ].value*. 

I'm starting to work on firmata and I will try and be able to contribute.

Thanks Jeff and Rick

closed 2016-01-16 by jguille2