firmata/firmata.js #109: make it easier and more obvious to use firmata with non-serial transports

soundanalogous, 2016-02-02:
Now that arduino Firmata has Ethernet, WiFi and soon BLE transports, it would be helpful to rethink how node-firmata is initialized. Perhaps this is as simple as updating the documentation to point to the respective transport objects that are already available.

rwaldron, 2016-02-02:
I'd prefer if this were a just a documentation update, ideally in the form of working examples of initialization alternatives.

cotejp, 2016-02-04:
I do not think it can be a simple documentation update. One of the first thing firmata does is check if it's running in the browser. If not, and if node-serialport is not available, it immediately throws an error...

Here's a specific example. In NW.js, my use case is this: `process.browser` is `false` but the error should not be thrown because NW.js can instead use the built-in `chrome.serial`. The code below works fine only if I comment out line 36 (`throw "Missing serialport dependency";`)

```
var io = require("firmata").Board;
var five = require("johnny-five");
var SerialPort = require("browser-serialport").SerialPort;

var board = new five.Board({
  io: new io(
    new SerialPort( "/dev/tty.usbserial-A9007KLo", { baudRate: 57600, bufferSize: 1 } )
  )
});
```

The reason I submitted [PR108](https://github.com/jgautier/firmata/pull/108) is so the code above could be simplified to:

```
var io = require("firmata").Board;
var five = require("johnny-five");
var board = new five.Board({ io: new io("/dev/tty.usbserial-A9007KLo") });
```

I submitted the additional [PR1028](https://github.com/rwaldron/johnny-five/pull/1028) for johnny-five so the example above can be further simplified to this:

```
var five = require("johnny-five");
var board = new five.Board();
```

While I can live without PR1028 (johnny-five), I do not see a way around PR108 (firmata). Does this make sense?

soundanalogous, 2016-02-04:
node-firmata can use non-serial transports, but it is not well know. This issue is about making it more well known and easier (where I agree with Rick that "easier" can be in the form of a few good examples). The way you use non-serial transports is to install an appropriate module. Here are three examples:

- [etherport](https://github.com/rwaldron/etherport) (for Firmata over Ethernet)
- [etherport-client](https://github.com/mwittig/etherport-client) (for Firmata over WiFi)
- [ble-serial](https://github.com/monteslu/ble-serial) (for Firmata over BLE)