firmata/firmata.js #112: How do I bypass FirmataEthernet if the network is down

dmastag, 2016-02-25:
Hi,

First of all, I love Firmata. We are using FirmataEthernet together with Node + Johnny Five to manage the lights, curtains, projector screens, etc in our Building. But we have a problem that we can't crack.

When the network is down in our building, we are not able to turn on & off the lights using regular light switches. This has not happened at all here but it is a use-case that could happen.

I was trying to add some digital input reads in the loop to check for switch changes but it somehow makes the jhonny-five unusable.

Could I get some direction on how to solve this problem?

Thanks Before
Julian

soundanalogous, 2016-02-25:
When you say "FirmataEthernet" do you mean "StandardFirmataEthernet" or some other firmware?

dmastag, 2016-02-25:
Yes, @soundanalogous. 
I meant StandardFirmataEthernet. Sorry if I wasn't clear before.

soundanalogous, 2016-02-25:
There is currently no way to fall back to a different transport so if the network is down then you will not be able to communicate with the Arduino using firmata.js / johnny-five when running the StandardFirmataEthernet firmware.

Your question is difficult to answer without some understanding of how everything is wired up.

dmastag, 2016-02-26:
Ok, I'll try to explain the use-case in points.
Please bear with me as I just started to learn about Arduino and Firmata last week, but I am keen to understand more and receive some positive feedback.

This is the current working Setup
- We are using Arduino connected to Digital Switches to control lights in a Building
- The Arduino is connected to an Ethernet Shield 
- The Arduino is using a Potentiometer and some Pins as a means for a static IP Address
- We are StandardFirmataEthernet on the Arduino to get it going
- The Arduino communicates to a Node Server using Johnny-Five + Etherport
- The Node Server tells the Arduino about the states of the Lights and which Digital Switches should affect which Lights
- Using Node we are able to control all the Lights in the Building
- We have 25 Arduinos communicating to a central Node Server with the above setup

The above is already working but we have the following struggles:
If the network is disconnected or a power failure happens in the building then all the light switches will stop working because the initial setup is done through Node

My thought to fix this is to add some code before `checkDigitalInputs();` to check the states of the digital switches, something in the lines of

```
if (digitalRead(switch1) == 1) {
    Serial.println("Switch 1 OFF");
    digitalWrite(led1, HIGH);
}
if (digitalRead(switch1) == 0) {
    Serial.println("Switch 1 ON");
    digitalWrite(led1, LOW);
}
```

But this has some problems in it:
* This overwrites everything coming from node/johnny-five when the ethernet connection is back on
* After a couple of loops arduino stops at `Firmata.available()` for 30 seconds and then continues to loop. So the Light Switches will not work during that time.

My approach is obviously wrong, I hope I can get some pointers on what the best approach would be.

dmastag, 2016-02-26:
Also about the logic for

```
if (digitalRead(switch1) == 1) {
    Serial.println("Switch 1 OFF");
    digitalWrite(led1, HIGH);
}
if (digitalRead(switch1) == 0) {
    Serial.println("Switch 1 ON");
    digitalWrite(led1, LOW);
}
```
I was thinking to put it in between some if statement, like this

```
if (!thisVariableToBeSetToTrueWhenConnectedToNode) {
    if (digitalRead(switch1) == 1) {
        Serial.println("Switch 1 OFF");
        digitalWrite(led1, HIGH);
    }
    if (digitalRead(switch1) == 0) {
        Serial.println("Switch 1 ON");
        digitalWrite(led1, LOW);
    }
}
```

For thisVariableToBeSetToTrueWhenConnectedToNode, I was hoping that I could change this variable through node so the code inside will be bypassed. Unless there is a function like Firmata.noconnection() that I can use.

soundanalogous, 2016-02-27:
This thread really belongs in the [firmata/arduino repo](https://github.com/firmata/arduino/issues) since it is an issue with the way Ethernet is handled on the firmware side.

There are a few issues that make implementing this challenging. The most significant issue is there is currently no way to immediately know when the Ethernet connection is down. This means data will continue to be buffered and when the buffer overflows will cause significant delays in code execution. It also means that when the connection is re-established you will get some portion of the previously buffered data. This aligns with what you described in a previous comment:

> But this has some problems in it:
- This overwrites everything coming from node/johnny-five when the ethernet connection is back on
- After a couple of loops arduino stops at Firmata.available() for 30 seconds and then continues to loop. So the Light Switches will not work during that time.

[This thread](https://github.com/firmata/ConfigurableFirmata/issues/37) goes into some of these issues and offers some potential solutions, one of which is [contingent on a future update to the Arduino Ethernet library](https://github.com/firmata/ConfigurableFirmata/issues/37#issuecomment-173007257).

However, there may be some things you can do in StandardFirmataEthernet:
- If you don't need to read switch state in johnny-five then you should set those pins to `PIN_MODE_IGNORE` in this [code block](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataEthernet/StandardFirmataEthernet.ino#L836-L845).
- Change [these lines]https://github.com/firmata/arduino/blob/master/examples/StandardFirmataEthernet/StandardFirmataEthernet.ino#L904-L907) at the bottom of the `loop()` function to:
```c++
// if this returns false, the connection is down
// (but this state is not known for some amount of time)
if (Ethernet.maintain()) {
  stream.maintain(Ethernet.localIP());
} else {
  // add your code for managing switches here
}
```

dmastag, 2016-02-28:
Thanks for the reply @soundanalogous, am going to try that out on Monday morning my time.