firmata/firmata.js #117: Browserify support

bcjordan, 2016-03-29:
Firmata currently fails to browserify because the `test/` directory is `.npmignore`d, but `test/MockSerialPort` gets conditionally `require()`d conditionally based on `global.IS_TEST_MODE`, resulting in the error:

`Error: Cannot find module '../test/MockSerialPort' from '/Users/brian/code/firmata-browserify/node_modules/firmata/lib'`

# Minimal repro repository

https://github.com/bcjordan/firmata-browserify-test

* `git clone https://github.com/bcjordan/firmata-browserify-test.git`
* `cd firmata-browserify-test`
* `npm run build`

Expected: browserify runs OK, maybe either including `MockFirmata` or requiring `MockFirmata` to be passed in by the tests?
Actual: 

```
npm run build

> firmata-browserify@1.0.0 build /Users/brian/code/firmata-browserify
> browserify main.js -o bundle.js

Error: Cannot find module '../test/MockSerialPort' from '/Users/brian/code/firmata-browserify/node_modules/firmata/lib'
    at /usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:55:21
    at load (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:69:43)
    at onex (/usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:92:31)
    at /usr/local/lib/node_modules/browserify/node_modules/resolve/lib/async.js:22:47
    at FSReqWrap.oncomplete (fs.js:95:15)

npm ERR! Darwin 14.5.0
npm ERR! argv "node" "/usr/local/bin/npm" "run" "build"
npm ERR! node v0.12.7
npm ERR! npm  v2.9.1
npm ERR! code ELIFECYCLE
npm ERR! firmata-browserify@1.0.0 build: `browserify main.js -o bundle.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the firmata-browserify@1.0.0 build script 'browserify main.js -o bundle.js'.
npm ERR! This is most likely a problem with the firmata-browserify package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     browserify main.js -o bundle.js
npm ERR! You can get their info via:
npm ERR!     npm owner ls firmata-browserify
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/brian/code/firmata-browserify/npm-debug.log
```

monteslu, 2016-03-29:
It's also an issue when trying to use webpack.

bcjordan, 2016-03-29:
I'd be up to take a stab at either:

1. Including `MockFirmata.js` in the npm package (moving it out of `test/` folder into `lib/`?)
2. Support passing in a given `SerialPort` module at initialization time, do that instead of conditionally swapping based on `globals.IS_TEST_MODE`

Any thoughts on better approach / other options?

bcjordan, 2016-03-29:
Here's a webpack test repo with the same repro steps (checkout, `npm install`, `npm run build`): https://github.com/bcjordan/firmata-webpack-test.git

Interestingly webpack yells a bunch of errors/warnings during build but ultimately builds out the bundle and appears to load OK.

monteslu, 2016-03-29:
haven't tested, but we might be able to just get away with adding something like this to the package.json:

```javascript
"browser": {
    "./test/MockSerialPort": false
}
```

rwaldron, 2016-03-30:
This was highly motivating—I needed to address some test mess anyway and this gave me the opportunity to clean that up, as well as resolve this issue.

rwaldron, 2016-03-30:
@monteslu @bcjordan I pinged you both on the PR. This is particularly notable: https://github.com/firmata/firmata.js/pull/118/files#diff-9288defa17db95568668c37e789a0b48

soundanalogous, 2016-06-19:
Has this been resolved?

rwaldron, 2016-06-20:
Yes, I should've closed this when that landed.

closed 2016-06-20 by rwaldron