firmata/firmata.js #119: Arduino Mega 2560 not working properly

ericogr, 2016-04-09:
The first execution after plug USB is ok but after open the program again (sometimes), the error appears. Tested in arduino mega and leonardo but _problem appears only in Arduino MEGA 2560_

**Sample js program:**
```
var Board = require('firmata');
Board.requestPort(function(error, port) {
	var board = new Board(port.comName);

	board.on('close', function() { console.info('close');});
	board.on('disconnect', function() { console.info('disconnect'); });
	board.on('open', function() { console.info('open');	});
	board.on('error', function(err) { console.info('error: ' + err); process.exit(-1);});
	board.on('ready', function() {
		var digital = 13;
		board.pinMode(digital, board.MODES.OUTPUT);
		board.digitalWrite(digital, 1);
		board.digitalWrite(digital, 0);
		board.serialClose(port);
	});
});
```
**output**
```

open
buffer.js:969
    throw new TypeError('value is out of bounds');
    ^

TypeError: value is out of bounds
    at checkInt (buffer.js:969:11)
    at Buffer.writeUInt8 (buffer.js:1017:5)
    at SYSEX_RESPONSE.(anonymous function) (/git/javascript/firmatajs/node_modules/firmata/lib/firmata.js:188:12)
    at Board.<anonymous> (/git/javascript/firmatajs/node_modules/firmata/lib/firmata.js:583:13)
    at emitOne (events.js:90:13)
    at SerialPort.emit (events.js:182:7)
    at Object.module.exports.raw [as parser] (/git/javascript/firmatajs/node_modules/serialport/parsers.js:8:13)
    at Object.SerialPort.opts.dataCallback (/git/javascript/firmatajs/node_modules/serialport/serialport.js:181:12)
    at SerialPortFactory.SerialPort._emitData (/git/javascript/firmatajs/node_modules/serialport/serialport.js:396:20)
    at afterRead (/git/javascript/firmatajs/node_modules/serialport/serialport.js:372:20)
<<<
```

**System**
- Linux Ubuntu 14.04.4 (3.13.0-83-generic)
- Firmata: 0.10.1
- Arduino Mega 2560
- StandardFirmata 2.5
- Node v5.10.1

**More info**
 Aparently, the data from serial port are inconsistent
_Added a console.log (firmata.js) to print buffer:_

```
SYSEX_RESPONSE[QUERY_FIRMWARE] = function(board) {
  var length = board.currentBuffer.length - 2;
  var buffer = new Buffer(Math.round((length - 4) / 2));
  var byte = 0;
  var offset = 0;

  console.log(board.currentBuffer); //<<<<<<<<<<<<<<<<<

  for (var i = 4; i < length; i += 2) {
    byte = (board.currentBuffer[i] & 0x7F) | ((board.currentBuffer[i + 1] & 0x7F) << 7);
    buffer.writeUInt8(byte, offset++);
  }
```

**Output:**
[ 240,  121,  2,  5,  83,  0,  116,  0,  97,  0,  110,  0,  100,  0,  97,  249,  2,  5,  240,  121,  2,  5,  83,  0,  116,  0,  97,  0,  110,  0,  100,  0,  7,  0,  114,  0,  100,  0,  70,  0,  105,  0,  114,  0,  109,  0,  97,  0,  116,  0,  97,
  0,  46,  0,  105,  0,  110,  0,  111,  0,  247 ]

soundanalogous, 2016-04-09:
Try updating to firmata.js 0.11.3. There were some parsing issues that were fixed.

ericogr, 2016-04-09:
Thanks!
now using 0.11.3, but the program hangs!

I added a log on "data" (firmata.js).

```
...
this.transport.on("data", function(data) {
    console.log(data); //<<<<<<<<<<<<
...
```

See these buffers:

```
<Buffer f9 02 05 f0 79 02 05 53 00 74 00 61 00 6e 00 64 00 61 00 72 00>
<Buffer f9 02 05 f0 79 02 05>
<Buffer 53 00 74 00 61 00 6e 00 64 00 61 00 72 00 64 00 46 00 69 00 72 00 6d 00>
<Buffer 61 00 74 00 61 00 2e 00 69 00 6e 00 6f 00 f7>

```
It hangs, not calling ready()

ericogr, 2016-04-09:
first attempt to fix #120

closed 2016-09-30 by ericogr