firmata/firmata.js #124: digitalRead EventEmitter leak

vicatcu, 2016-05-12:
I'm pretty sure I'm missing something fundamental, but I'm stuck so here goes. I'm polling a button like this (gBoard is my firmata object):

let button_pressed = false;
let startButtonPollingLoop = () => {
  gBoard.digitalRead(BUTTON_PIN, function(value){
    button_pressed = (value == 0) ? true : false;
    setTimeout(startButtonPollingLoop, 10);
  });
};

... pretty quickly I get this in my node console:

> (node) warning: possible EventEmitter memory leak detected. 11 digital-read-16 listeners added. Use emitter.setMaxListeners() to increase limit.

This is obviously happening, on review of the firmata.js code, because of [this](https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L887). Clearly I'm doing it wrong, but I don't see how to do it right. Can anyone help?

soundanalogous, 2016-05-12:
digitalRead is event driven. There is no need to poll. The callback function will be called whenever the button is pressed.

vicatcu, 2016-05-12:
@soundanalogous thanks, yes I just experimentally figured that out as well, that's kind of awesome. Quite different from what digitalRead does in Arduino-land though. Might it be a good idea to guard against (maybe even notify via in the console) when adding multiple listeners to the same pin? Or maybe just update the README to make that a bit clearer. Anyway, I'll close this issue.

closed 2016-05-12 by vicatcu

soundanalogous, 2016-05-12:
Arduino is procedural, Node is event driven. It probably needs to be made more clear in the documents, but really it's the nature of JavaScript vs c/c++.

rwaldron, 2016-05-13:
> Might it be a good idea to guard against (maybe even notify via in the console) when adding multiple listeners to the same pin

That should be left libraries that relies on Firmata.js, not baked in explicitly. I agree with Jeff, re: docs fix