firmata/firmata.js #127: Why do digitalRead/analogRead turn reporting pins on?

PKGeorgiev, 2016-06-09:
I'm really confused of how digitalRead/analogRead work.
In fact these two turn pin reporting on. In case of analogRead it floods host app with data.
On the other hand, digitalRead's callback is only called on change. May be names are confusing. 
One would expect simple flow:

digitalRead > sendPaketToArduino > digitalRead(port) /Arduino side/ > sendDigitalResponse > invokeCallback

The same applies to analogRead. Just single reads.
Now digitalRead is: "reportDigitalPortChanges" and analogRead is: "turnOnPinReporting".

Also, digitalWrite/analogWrite should execute callback so there is feedback from the write requests. Now it is like "fire and forget". Current implementations assume digitalWrite/analogWrite will succeed. But that's not always the case. 
Following NodeJs recommendations every function should execute a callback with an error var as first parameter.

soundanalogous, 2016-06-09:
Firmata does not report digital or analog pin changes unless pin reporting is enabled. This is to prevent flooding the host with unwanted data. In some firmata host libraries it is up to the user to enable pin reporting separately, but for firmata.js a decision was made to simplify this for the user and enable reporting only when a digital read or analog read request is sent.

I agree that `digitalRead` and `analogRead` are misleading choices of method names for wrapping the `DIGITAL_MESSAGE` or `ANALOG_MESSAGE` in any Firmata host library because they do not function like the Arduino/Wiring equivalent. They should really be named something like `startDigitalRead` and `startAnalogRead` since that is what the actual behavior is. They could then be complimented with `stopDigitalRead` and `stopAnalogRead`. There is an addendum to the Firmata protocol that would add the ability for a true digitalRead or analogRead (as in read the specified pin only once) but it is a less efficient approach than using the reporting methods.

Regarding digitalWrite and analogWrite, the Firmata protocol does not ACK writes. Theoretically you could using the pin state query, but that is another round trip from the host to the firmware and back.

soundanalogous, 2016-06-09:
@rwaldron FYI about method names. It may be worth making a change like this at some point and deprecating digitalRead and analogRead. Or at least better explaining how those methods actually function. I'm starting to see a wave of people who started with Arduino and are now exploring Firmata hosts (firmata.js, johnny-five), etc. They are confused by the similar Wiring API but different functionality.

PKGeorgiev, 2016-06-09:
@soundanalogous thank you for your fast response. Better explanation would be very helpful. I spent a few hours trying to understand "what happens here".
My case is simple: I look for strong protocol for communication between host and Arduino. Firmata looks promising.
I come from a "transactional world" and I expected to see request/response like style APIs. Adding single query/response to the library will be very helpful. I get the "big picture", but I was confused by method naming and different behavior on each platform.

troywweber7, 2016-09-21:
I'd like to know how to get a single read out of an analog pin every 1 second. I can't see any way to stop the analog reading, and as mentioned, it just floods the serial line unnecessarily. Ideas?

rwaldron, 2016-09-22:
You can change speed of the polling interval: 

```js
board.setSamplingInterval(1000);
```

troywweber7, 2016-09-22:
> You can change speed of the polling interval:
>
> board.setSamplingInterval(1000);

Excellent, thank you! I presume there is a command to stop the reading
altogether and I probably just need to look in the code harder to find it?

troywweber7, 2016-09-22:
I think I found it. I presume I can just use
```javascript
board.reportDigitalPin(14,0);
```
To turn off the reporting on digital pin 14.

bsefton, 2016-10-02:
I am having a problem using analogRead with firmata. I am running firmata on a Yun shield runnng OpenWRT Linux with node.js and firmata. Everything is installed correctly, and I was able to create an LED blink program and add web based on/off to it based on values passed through the url. 

I am now trying to do an analogRead and am getting nowhere. THis doesnt seem to work despite trying lots of attempts.  Does anyone have, or know where I can get a javascript code snippet which uses firmata and shows how to do an analogRead? I am trying to read a dht11 temp and humidity sensor and display the output via console.log as a basic test.

THanks

soundanalogous, 2016-10-02:
DHT11 is digital not analog so that may be part of the issue. There is actually no way to read that sensor directly with firmata.js since it uses a protocol that is not supported by Firmata. You have to use an I2C backpack or modified version of Firmata that provides DHT11 support.

Regarding how to read an analog input with firmata.js, you simply do this:

```javascript
var Board = require("firmata");
Board.requestPort(function(error, port) {
  if (error) {
    console.log(error);
    return;
  }
  var board = new Board(port.comName);
  board.on("ready", function() {
    this.pinMode(0, this.MODES.ANALOG);
    // read pin A0
    this.analogRead(0, function(value) {
      console.log(value);
    });
  });
});
```

bsefton, 2016-10-02:
Thanks, I got it working with the following code in case anyone is interested. You are right the DHT11 is digital and gives a numeric value. Adding the board.setSamplingInterval(1000); line reduces the output to one value per second. The specification of the serial port like this is the only way I could find to make this work on a Dragino (yun shield) on an Uno. It is configured like this tutorial (however I did not install and am not using cyclon.js) https://cylonjs.com/documentation/platforms/yun/
This code may be only relevant for those runing yun or dragino liux boards, but it works.

var serialport = require('serialport');
var sp = new serialport.SerialPort('/dev/ttyATH0', {baudrate:57600});
var firmata = require('firmata');
var board = new firmata.Board(sp, function(err) {
  if (err) {
      console.log(err);
      board.reset();
      return;
   }
 console.log('connected...');
 console.log('board.firmware: ', board.firmware);
   board.setSamplingInterval(1000);
   board.analogRead(0, function(val) {
   console.log(val);
                    });
});

soundanalogous, 2016-10-02:
That cylonjs article references a really old version of StandardFirmata (from 2011). This newer one should be used instead: https://github.com/firmata/StandardFirmataYun.

gkzsolt, 2016-10-29:
I'd like to return to @PKGeorgiev 's opening comment:
> I'm really confused of how digitalRead/analogRead work.
In fact these two turn pin reporting on. In case of analogRead it floods host app with data.
On the other hand, digitalRead's callback is only called on value change. In Arduino *Read() means "single read". May be names are confusing.
One would expect simple flow:
digitalRead > sendPaketToArduino > digitalRead(port) /Arduino side/ > sendDigitalResponse > invokeCallback
The same applies to analogRead.
Now digitalRead is just: "reportDigitalPortChanges" and analogRead is: "turnOnAnalogPinReporting". This makes NodeJs app's flow much more complicated. For example - a trivial task - how can we read single port value? 

I completely agree. There are 2 issues here:

(1) reading a pin
(2) reporting pin value changes

Reading a pin (1) is done by digitalRead() or analogRead() on the Arduino and provide a simple direct pin read. If we are interested only in pin changes (2) then we'll use a function or library which stores a pin's state, reads the pin regularly and reports the state change. For example, the Bounce library's rose() function returns true if signal transitions from low to high (and fell() does the reverse). 
(1) and (2) are 2 different functions: the second is the derivative of the first with respect to time.

Now Firmata seems to have implemented only the derivative while keeping the original function name. This is extremely confusing. Some library providers followed this. Johnny-Five board API informs us that:

> digitalRead(pin, handler(value))
Register a handler to be called whenever the board reports the value (0 or 1) of the specified digital pin.

No, please. This should be called `onPinChange`, `onDigitalPinChange`, or something similar, but in neither case `digitalRead`. Neither can I agree with `startDigitalRead`, because it is not a read what we'll get in the callback. The handler will be called only if the pin state changes (and not on all pin reads), and this should be reflected in the name, to be clear. That's about the naming.

On the other hand, I took for granted that a communication library can read pin states, glancing through the API calls reassured me on that so it came as a big surprise that it cannot. Direct digital- and analogRead calls are basic and I'd expect Firmata to do also this. To infer a pin state from its derivative (by storing an initial value and then integrating the changes) is very costly for such a trivial task.

So I'd expect a `digitalRead(pin, callback)` to read the pin, then send the value back to notify a registered once listener for the callback (of course the naming is problematic due to legacy issues, so some plan should be made.)

A similar behaviour would be logical for `analogRead`. This seems to do a real analog read ;), so the function itself and not the derivative - but only if reportAnalog was enabled, when it is reporting the value steadily at the set frequency rate. Then reportAnalog should be stopped by the client. Well, this matches well the way I'd use it, so I'll not complain here ;) A question here: is `analogReference()` implemented in Firmata? (it may be an Arduino-specific call, but how to handle this?)

@PKGeorgiev, I don't think digitalWrite/analogWrite should execute callback so there is feedback from the write requests. There is no feedback from the Arduino itself, so this would hardly be possible. You can do a read afterwards (if digitalRead() was implemented ;) if you want to be sure the command has completed (at least as regarding digitalWrite. I don't know much about analogWrite.)

I hope these make sense.

soundanalogous, 2016-10-30:
The [Firmata protocol](https://github.com/firmata/protocol) makes no claim to being a wrapper for the Arduino/Wiring api. It actually makes no claim to being exclusive to Arduino either. It is a general protocol for communicating with microcontrollers that could theoretically be implemented on any microcontroller platform.

There is also no requirement in the Firmata protocol that analog pin reporting or digital port reporting should be enabled automatically when sending a read request.

> A question here: is analogReference() implemented in Firmata?

Not currently. I had submitted [a proposal](https://github.com/firmata/protocol/pull/8) a couple of years ago, then ran into a technical issue with the implementation (it was difficult to handle the differences between the AVR and SAM architectures but Arduino has since added constants to make that easier to deal with).

soundanalogous, 2016-10-30:
> Direct digital- and analogRead calls are basic and I'd expect Firmata to do also this.

Please submit a proposal to add such functionality: https://github.com/firmata/protocol/issues.

I could see a message to query an analog pin to be useful. However I don't see any real good use cases for a way to query a digital pin value, mostly because of the latency involved in the round trip transmissions. However there are real uses cases other than a purely semantic advantage I could be convinced otherwise.

gkzsolt, 2016-10-30:
What about the naming? How are we supposed to get the value of a pin?

soundanalogous, 2016-10-30:
With firmata.js, you can get the pin value at any time like this:

```javascript
// enable reporting for pin D2 (actually enables the entire port that pin D2 belongs to)
board.reportDigitialPin(2, 1);

// get the current value of this pin
// this is updated once when reporting is enabled for the port
// (actually all pins in that port are updated) and any time the pin
// changes on the Arduino, as quickly as that can be reported.
board.pins(2).value;

// for an analog pin (A0)
board.reportAnalogPin(0);

// if you know the digital pin number equivalent for the analog pin (my vary by board)
board.pins(14).value;
// if you don't know the digital pin number
board.pins[board.analogPins[0]].value;
```

What is missing however is something like this:

```javascript
board.pins[2].on("change", callback);
```

soundanalogous, 2016-10-30:
The problem in implementing `board.pins[2].on("change", callback);` however is that the object stored in `board.pins[pinNum]` is not an event emitter. This could be changed by refactoring to create a separate `pin` module that extends `EventEmitter`.

soundanalogous, 2016-10-30:
The firmata.js documentation of the digitalRead and analogRead methods also needs to be updated because it is currently misleading.

soundanalogous, 2016-10-30:
This update to the firmata.js documentation should help eliminate the confusion. I agree that the `analogRead` and `digitalRead` method names are not ideal. Please open a separate issue to address that specifically.

https://github.com/firmata/firmata.js/pull/138

rwaldron, 2016-10-31:
> I agree that the analogRead and digitalRead method names are not ideal. Please open a separate issue to address that specifically.

Honestly, I think that ship sailed a long time ago :|

> https://github.com/soundanalogous/firmata/blob/8b8d064494a6f3c3a621f607b3d11fccc06f03d3/readme.md

Read and 👍

PKGeorgiev, 2016-10-31:
Guys, thank you for updating the documentation! It's clearer now :)
Also thanks to @garrows about his detailed comment(s).

As a home automation guy my main concern is about getting feedback from invoked commands. Let me give you a real world example: you control a power socket with Arduino (digital pin) via Frimata.js. You press the button of your remote control, a packet is send to Arduino and the socket is turned on or off - eventually. A digital pin report packet is sent back to server independently of your command. You cannot relate these two (the command packet and the digital report event). Now imagine you plugged a heater into  that power socket and you want to turn it off. You send command, it may be successful. But what if it wasn't? You may think you turned off the heater, but it's actually not. You don't know your command status. This is very dangerous when it comes to home automation. 

On the other hand NodeJS offers nice callback/error mechanism. It may be utilized to implement some request/response logic. This way you can easily relate the command and its status. In ModBus there is a notion of transaction ID. Another option would be to be able to tag packets. May be this would be considered as a feature request.

I think request/response roundtrip is not a major issue given that reporting with the default sampling interval of 19ms is already flooding the line when doing analog readings.

Arduino is serious, Firmata is serious, Projects are getting more and more serious and complex. Think about it.

soundanalogous, 2016-10-31:
Firmata is based on midi (and still largely midi-compliant) and midi does not ACK messages, therefore Firmata does not ACK messages either. It is something I could consider for Firmata 3.0 if there is sufficient interest in taking Firmata in this direction so please open a thread in the [firmata/protocol repo](https://github.com/firmata/protocol) if that is something you are interested in.

For now the best thing you can do is call `board.queryPinState(pinNum, callback)` immediately after `board.digitalWrite(pinNum, value)` and check that `board.pins[pinNum].state === valueYouWrote`.

> Arduino is serious, Firmata is serious, Projects are getting more and more serious and complex. 

Arduino has a full-time engineering team plus a large developer community. Firmata (the protocol + arduino implementation) has a very small developer community adding up to just a handful of hours per month.

soundanalogous, 2016-10-31:
I'm going to close this issue since I think the updates to the documentation were sufficient. Any other issues that were raised in this thread should be opened in their respective repositories (some belong in firmata/protocol for example).

closed 2016-10-31 by soundanalogous

gkzsolt, 2016-10-31:
Hi guys,

Sorry to write after the issue was closed, I didn't get a chance to answer.

> The Firmata protocol makes no claim to being a wrapper for the Arduino/Wiring api.

May be, but it borrows some API call names from it with different semantics; and this will create all kind of confusions for newcomers.

> This update to the firmata.js documentation should help eliminate the confusion. [...]
https://github.com/soundanalogous/firmata/blob/8b8d064494a6f3c3a621f607b3d11fccc06f03d3/readme.md

Thanks very much for this! But more specifically, thanks the pointer to that piece of documentation! I did not know about. I started from the [Arduino libraries page](https://www.arduino.cc/en/Reference/Firmata), that linked to the [Firmata protocol page](https://github.com/firmata/protocol), and this seemed so complete (enumerating all client libraries, etc.) that I didn't think to look further. I downloaded the source and that was it.
(It is not always enough to describe the protocol, a guide to the most common use cases is welcome. Yes, I know that this is implementation, but the client libraries also are.)
So a note/link to that firmata.js documentation would be tremendously useful! Who should I ask for this?

@rwaldron: 
>> I agree that the analogRead and digitalRead method names are not ideal. Please open a separate issue to address that specifically.
> Honestly, I think that ship sailed a long time ago :|

May be, but digitalRead() is still very confusing on johnny-five and will puzzle every newcomer. Also, it didn't work for me for digital output pins. I don't know why that happened...

@PKGeorgiev, you don't have always feedback from the issued commands. But you can ask for it: if you read the power socket state after you sent the "turn off" command, say repeatedly with a timeout, you will see if it really turned off. If that didn't happen in an hour, the home automation can call the firefighters ;) Actually you cannot have a direct reliable feedback from any serial communication, because of its asynchronous nature (the only solution seems to be tagging each request/response, but this won't help much here because of the latency.)

@soundanalogous, thanks again for clarification, help and updating the documentation.

rwaldron, 2016-10-31:
@gkzsolt 

> May be, but digitalRead() is still very confusing on johnny-five and will puzzle every newcomer. Also, it didn't work for me for digital output pins. I don't know why that happened...

J5's `Board.prototype.digitalRead` is just a pass through to firmata.js's method of the same name. 

> Also, it didn't work for me for digital output pins.

That's correct, you have to call `board.pinMode(pin, board.MODES.INPUT)` to have a pin treated like an input. 

@PKGeorgiev 

> You send command, it may be successful. But what if it wasn't? You may think you turned off the heater, but it's actually not. You don't know your command status. This is very dangerous when it comes to home automation. 

This can be achieved with a sensor on the heater.

PKGeorgiev, 2016-10-31:
@gkzsolt a command should always get feedback either success or error irrespective of the physical medium. This way your program can make dicisions. That's my point. Yes, I could read pin state, but it is not safe, because state change packet may not be received by the server side (consider noisy rf communication) which makes Firmata unreliable protocol.   I'll propose an enhancement to the firmata protocol repo.

gkzsolt, 2016-10-31:
> That's correct, you have to call board.pinMode(pin, board.MODES.INPUT) to have a pin treated like an input. 

So it means that you cannot listen to OUTPUT pin state changes? I didn't know that Arduino digitalRead() is limited to INPUT pins, but yeah, this is not Wiring...

soundanalogous, 2016-10-31:
>The Firmata protocol makes no claim to being a wrapper for the Arduino/Wiring api.
>>May be, but it borrows some API call names from it with different semantics; and this will create all kind of confusions for newcomers.

I think there is some confusion here between the [Firmata protocol](https://github.com/firmata/protocol) and [Arduino implementation](https://github.com/firmata/arduino) and [firmata.js](https://github.com/firmata/firmata.js). These are 3 separate projects. The Firmata protocol borrows absolutely no API call names from Arduino/Wiring nor does it enforce any kind of method naming convention. firmata.js does borrow some API call names from Arduino/Wiring, but firmata.js is nodejs implementation of the Firmata protocol, it is not a protocol itself, nor is it a documentation of the protocol.

gkzsolt, 2016-10-31:
Oh, sorry, you are right and this is very clear, thanks. I found also the link to the firmata.js implementation, it turned out to be https://github.com/jgautier/firmata.

rwaldron, 2016-11-03:
>  it turned out to be https://github.com/jgautier/firmata.

That's _this_ project ;)