firmata/firmata.js #131: Unable to Use Software Serial on Arduino Yun

stardawg, 2016-09-02:
I am trying to control a roomba with an arduino yun using software serial on pins 10 and 11.  However, board.SERIAL_PORT_IDs is undefined, so I cannot configure the software serial port.  On the mcu I am running standard firmata sketch.

   var board = new firmata.Board(myPort, function (err) {
       if (err) {
           console.log(err);
           board.reset();
           return;
       } else {
           console.log('connected...');

           console.log('board.firmware: ', board.firmware);
               console.log("READY");
               var SW_SERIAL0 = board.SERIAL_PORT_IDs.SW_SERIAL1;
               var maxBytesToRead = 4;

               board.serialConfig({
                   portId: SW_SERIAL0,
                   baud: 19200,
                   rxPin: 10,
                   txPin: 11
               });

Thanks,
Kevin

[image]

soundanalogous, 2016-09-02:
I'll look into the SW serial issue this weekend (I'm traveling now). However with the Yun you should use HW serial instead. Connect to pins 1 & 2 and use HW_SERIAL1. It will work much better.

Also make sure you have StandardFirmataPlus uploaded to your board, that is the variant that includes Serial support. StandardFirmata does not.

soundanalogous, 2016-09-02:
Oh, I thought at first you were talking about Arduino firmata (the fact that the node version was also named "firmata" is way too confusing). Hummm, make sure you have the latest version of firmata.js installed. Also, make sure you listen for the "ready" event:

```javascript
var Board = require("firmata");
Board.requestPort(function(error, port) {
  if (error) {
    console.log(error);
    return;
  }
  var board = new Board(port.comName);
  board.on("ready", function() {
    console.log("READY");
    console.log(
      board.firmware.name + "-" +
      board.firmware.version.major + "." +
      board.firmware.version.minor
    );

    var SW_SERIAL = board.SERIAL_PORT_IDs.SW_SERIAL0;
    board.serialConfig({
      portId: SW_SERIAL,
      baud: 19200,
      rxPin: 10,
      txPin: 11
    });

    board.serialRead(SW_SERIAL, function(data) {
      console.log(new Buffer(data).toString("ascii"));
    });
  });
});
```

But again with the Yun you can use HW serial on pins 0 and 1 instead which will perform much better.

stardawg, 2016-09-02:
Thanks for the quick responses!

I updated firmata to the latest version and I am now getting past the undefined error.  I am now able to communicate with the mcu from the linux board via Serial1 and StandardFirmataPlus (modified for yun).  I can confirm this by making the onboard led blink.  I am still unable to get the roomba to respond to commands via SoftwareSerial.  Attempting to use the hardware serial on pins 1 & 2 makes the program hang when trying to create the firmata board.  I suspect this is because I am using these pins for the communication between the linux board and the mcu.

Again, thanks for the help!

soundanalogous, 2016-09-02:
HW serial is actually pins 0 and 1, but yeah if you are connecting the Yun to another board then you're probably using those pins already (note however that on the Yun, the USB serial != pins 0 and 1 like on the Arduino Uno). There are so many potential issues with SoftwareSerial that it's really difficult to diagnose the issue (things like if certain digital pins are used, if certain timers are used, etc all affect how well SoftwareSerial works if it works at all).

soundanalogous, 2016-09-02:
Also, the Yun has an on-board linux computer so in effect you have a linux board that is bridged to an ATMega that is bridged to another linux computer. That may not play well.

stardawg, 2016-09-03:
my firmata.js code is running on the yun's on-board linux using Serial1 to communicate with the mcu.  The idea is to add web sockets so I can remote control the roomba through the yun over the internet.  I can control the roomba on pins 10 / 11 using a SoftwareSerial sketch directly on the mcu, but I prefer to control from js on the on-board linux.

[image]

soundanalogous, 2016-09-03:
Does switching `board.SERIAL_PORT_IDs.SW_SERIAL1` to `board.SERIAL_PORT_IDs.SW_SERIAL0` make any difference? Firmata allocates the SW serial ports in order(starting with SW_SERIAL0). It's different than HW_SERIAL where the constants are mapped to specific UARTs on the MCU (where HW_SERIAL1 = Serial1, HW_SERIAL2 = Serial2, etc).

stardawg, 2016-09-03:
Negative.

It's very flaky.  Sometimes the commands relay to the roomba, but most of the time not.  Also, I can sometimes get data back from the roomba, but it's all garbled and not in the expected format.

soundanalogous, 2016-09-03:
When data is sent over Firmata, bytes are split, are you recombining them as they are received?

soundanalogous, 2016-09-03:
Actually [they are combined in firmata.js](https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L378-L383). I should have know, I wrote that code. My memory fails me.

soundanalogous, 2016-09-03:
Can you post the Arduino code you said you wrote to control the Roomba directly from the MCU. That will help me understand if there is something you are doing there that I did not anticipate when adding SW serial support to Firmata.

soundanalogous, 2016-09-03:
Also, please post your full J5 code as well.

stardawg, 2016-09-08:
this has been a busy week.  I will post code within the next couple of days.

soundanalogous, 2016-09-08:
I'll be out of town for the next 2 weeks so I won't have time to look at it until after 9/25 so no rush.

stardawg, 2016-09-09:
HelloNancyRoomba.ino
====================================================
#include <SoftwareSerial.h>

 boolean isRoombaInitialized = false;

SoftwareSerial roombaSerial(10, 11); // RX, TX (instance of Serial1 object)

// Initialization Code ============================================
void initializeRoomba()
{
  if(!isRoombaInitialized)
  {
      roombaSerial.write(128);  //start (goes to Passive)
      delay(20);  //delay 20 milliseconds after state change
      roombaSerial.write(130);  //Command mode - goes to safe
      delay(20);  //delay 20 milliseconds after state change
      Serial1.println("Roomba Initialized (start, command)"); 
      isRoombaInitialized = true;
    
  }
}

void setup() {
  pinMode(13, OUTPUT);
  // Open Serial1 communications and wait for port to open:
  Serial1.begin(9600);
  Serial1.println("Menu 1");  //tell monitor we are alive

  // set the data rate for the SoftwareSerial1 port
   roombaSerial.begin(19200); //SoftSerial1 works at 19200!!
   delay(1000);
   initializeRoomba();
   
   blink();
}

void loop() {
  displayHelloNancy();
  
//  if (roombaSerial.available())   //First order of business: listen to Roomba
//  {
//    blink();
//    Serial1.println(roombaSerial.read());   //writes to USB input from soft Serial1
//  }
  while(roombaSerial.available())
  {
    Serial1.println(roombaSerial.read()); 
  }
}

void blink()
{
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(200);               // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(200);               
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(200);               // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
}

void displayHelloNancy() 
{
   byte command = 164;
    byte bH = 72;
    byte bE = 69;
    byte bL = 76;
    byte bO = 79;
    byte bN = 78;
[... 175 lines cut ...]
    board.serialConfig({
        portId: roombaSerial,
        baud: 19200,
        rxPin: 10,
        txPin: 11
    });

    // log serial pin numbers
    for (var pin in board.pins) {
        var modes = board.pins[pin].supportedModes;
        for (var mode in modes) {
            if (modes[mode] === board.MODES.SERIAL) {
                console.log("serial pin: " + pin);
            }
        }
    }

    board.serialRead(roombaSerial, maxBytesToRead, function (data) {
        console.log('from roomba: ' + new Buffer(data).toString("ascii"));
    });
    while (true) {
        displayHelloNancy();
        sleep(500);
    }
});

stardawg, 2016-09-09:
output from node console on yun on-board linux:
=======================================================
root@Arduino:~/node# node helloNancyRoomba
[2016-09-09 05:24:01.469] [INFO] console - opening comm port:  /dev/ttyATH0
[2016-09-09 05:24:02.799] [INFO] console - creating board...
[2016-09-09 05:24:08.061] [INFO] console - ready!
[2016-09-09 05:24:08.070] [INFO] console - connected...
[2016-09-09 05:24:08.078] [INFO] console - board.firmware:  { name: 'StandardFirmataPlus.ino',
  version: { major: 2, minor: 5 } }
[2016-09-09 05:24:08.132] [INFO] console - READY
[2016-09-09 05:24:08.139] [INFO] console - setting up software serial port for roomba comm...
[2016-09-09 05:24:08.150] [INFO] console - serial pin: 0
[2016-09-09 05:24:08.157] [INFO] console - serial pin: 1
[2016-09-09 05:24:08.180] [INFO] console - say hello...
[2016-09-09 05:24:08.210] [INFO] console - initializing roomba...
[2016-09-09 05:24:08.220] [INFO] console - zzz...
[2016-09-09 05:24:08.248] [INFO] console - zzz...
[2016-09-09 05:24:08.274] [INFO] console - Roomba Initialized!
[2016-09-09 05:24:08.280] [INFO] console - sending bytes: 16
[2016-09-09 05:24:08.286] [INFO] console - sending message to roomba...
[2016-09-09 05:24:08.293] [INFO] console - zzz...
[2016-09-09 05:24:08.499] [INFO] console - sending message to roomba...
[2016-09-09 05:24:08.505] [INFO] console - zzz...
[2016-09-09 05:24:08.710] [INFO] console - sending message to roomba...
[2016-09-09 05:24:08.716] [INFO] console - zzz...
[2016-09-09 05:24:08.923] [INFO] console - sending message to roomba...
[2016-09-09 05:24:08.929] [INFO] console - zzz...
[2016-09-09 05:24:09.134] [INFO] console - sending message to roomba...
[2016-09-09 05:24:09.140] [INFO] console - zzz...
[2016-09-09 05:24:09.346] [INFO] console - sending message to roomba...
[2016-09-09 05:24:09.352] [INFO] console - zzz...
[2016-09-09 05:24:09.559] [INFO] console - sending message to roomba...
[2016-09-09 05:24:09.565] [INFO] console - zzz...
[2016-09-09 05:24:09.770] [INFO] console - sending message to roomba...
[2016-09-09 05:24:09.776] [INFO] console - zzz...
[2016-09-09 05:24:09.983] [INFO] console - sending message to roomba...
[2016-09-09 05:24:09.989] [INFO] console - zzz...
[2016-09-09 05:24:10.195] [INFO] console - sending message to roomba...
[2016-09-09 05:24:10.201] [INFO] console - zzz...
[2016-09-09 05:24:10.407] [INFO] console - sending message to roomba...
[2016-09-09 05:24:10.413] [INFO] console - zzz...
[2016-09-09 05:24:10.620] [INFO] console - sending message to roomba...
[2016-09-09 05:24:10.626] [INFO] console - zzz...
[2016-09-09 05:24:10.832] [INFO] console - sending message to roomba...
[2016-09-09 05:24:10.838] [INFO] console - zzz...
[2016-09-09 05:24:11.044] [INFO] console - sending message to roomba...
[2016-09-09 05:24:11.050] [INFO] console - zzz...
[2016-09-09 05:24:11.256] [INFO] console - sending message to roomba...
[2016-09-09 05:24:11.262] [INFO] console - zzz...
[2016-09-09 05:24:11.468] [INFO] console - sending message to roomba...
[2016-09-09 05:24:11.474] [INFO] console - zzz...
[2016-09-09 05:24:11.680] [INFO] console - zzz...
====================================================

rwaldron, 2016-09-10:
Note: I edited that to add back-tick fences for syntax highlighting

stardawg, 2016-09-13:
I found some issues with the js file.  Below is a new version where I can see commands flow to the roomba.  However, it causes some weird behavior w/ the firmata sketch.  It appears that commands somehow get batched in the sketch and transferred to the roomba all at once instead of individually.

'''js
var log4js = require('log4js');
log4js.configure({
    appenders: [
        { type: 'console' },
        {
            "type": "dateFile",
            "filename": "logs/roombaBot.log",
            "pattern": "-yyyy-MM-dd",
            "alwaysIncludePattern": false
        },
    ],
    replaceConsole: true
});
var logger = log4js.getLogger();

var ledPin = 13;
var isHigh = false;
var firmata = require('firmata');
serialport = require('serialport');// include the library
SerialPort = serialport.SerialPort; // make a local instance of it
portName = '/dev/ttyATH0';
var board;
var roombaSerial;
var isRoombaInitialized = false;

console.log('opening comm port:  %s', portName);
var myPort = new SerialPort(portName, {
    baudRate: 57600,
});

function delay(millis) {
    var start = Date.now();
    var end = start + millis;
    while (Date.now() < end) { }
    console.log('slept: ' + (Date.now() - start));
}

function initializeRoomba() {
    if (!isRoombaInitialized) {
        console.log('initializing roomba...');
        board.serialWrite(roombaSerial, 128);  //start (goes to Passive)
        delay(20);  //delay 20 milliseconds after state change
        //board.serialFlush(roombaSerial);
        board.serialWrite(roombaSerial, 130);  //Command mode - goes to safe
        delay(20);  //delay 20 milliseconds after state change
        console.log("Roomba Initialized!");
        isRoombaInitialized = true;
    }
}

function scrollMessage(inputMessage, charDelay) {
    var command = 164;
    var blank = 32;
    var t1 = [command, blank, blank, blank, blank];
    var t2 = [];
    initializeRoomba();
    var message = inputMessage.split('');
    message.push(' ');
    message.push(' ');
    message.push(' ');
    message.push(' ');
    console.log('sending command to roomba:= ' + t1.toString());
    board.serialWrite(roombaSerial, t1);
    for (var i = 0; i < message.length; i++) {
        t2.push(t1.shift());
        t1.shift();
        t2.push(t1.shift());
        t2.push(t1.shift());
        t2.push(t1.shift());
        console.log('pushing char to output array: ' + message[i]);
        t2.push(message[i].charCodeAt(0));
        console.log('sending command to roomba:= ' + t2.toString());
        board.serialWrite(roombaSerial, t2);
        t1 = t2;
        delay(charDelay);
    }

}

function toggleLed() {
    if (isHigh) {
        board.digitalWrite(ledPin, board.HIGH);
    } else {
        board.digitalWrite(ledPin, board.LOW);
    }
    isHigh = !isHigh;
}

function blink(blinkDelay, blinkIterations)
{
    for (var i = 0; i < blinkIterations; i++) {
        toggleLed();
        delay(blinkDelay);
    }
    console.log('led is on: ' + !isHigh);
}

function main() {
    toggleLed();
    var message = 'my name is Mo';
    var charDelay = 200;
    var blinkDelay = 100;
    var blinkIterations = 4;
    var loopInterval = (message.length * charDelay) + (4 * charDelay) + (blinkIterations * blinkDelay) + 1000;
    console.log('loop interval: ' + loopInterval);
    setInterval(function () {
        var start = Date.now();
        blink(blinkDelay, blinkIterations);
        scrollMessage(message, charDelay);
        console.log('main: ' + (Date.now() - start));
        console.log('loop interval: ' + loopInterval);
    });
}

console.log('creating board...');
board = new firmata.Board(myPort, function (err) {
  if (err) {
    console.log(err);
    board.reset();   
    return;       
  }
});

board.on('ready', function () {
    console.log('connected...');
    console.log('board.firmware: ', board.firmware);

    board.pinMode(ledPin, board.MODES.OUTPUT);
    roombaSerial = board.SERIAL_PORT_IDs.SW_SERIAL0;
    console.log('roomba serial: ' + roombaSerial);

    board.serialConfig({
        portId: roombaSerial,
        baud: 19200,
        rxPin: 10,
        txPin: 11
    });

    console.log('rommba serial configuration completed.')
    //while (true) {
        main();
    //}
});
'''

stardawg, 2016-09-13:
I've attempted to narrow down the issue and have stripped software serial out of my js code leaving only blink functionality.  The code below should blink the led very slowly, 1s on 1s off.  It should blink 4 times in each iteration (loop interval of main).  However, what I see is the led come on for maybe 100ms once per loop interval at the end of main.  Code and output are below:

```js
var log4js = require('log4js');
log4js.configure({
    appenders: [
        { type: 'console' },
        {
            "type": "dateFile",
            "filename": "logs/roombaBot.log",
            "pattern": "-yyyy-MM-dd",
            "alwaysIncludePattern": false
        },
    ],
    replaceConsole: true
});
var logger = log4js.getLogger();

var ledPin = 13;
var isHigh = false;
var firmata = require('firmata');
serialport = require('serialport');// include the library
SerialPort = serialport.SerialPort; // make a local instance of it
portName = '/dev/ttyATH0';
var board;

console.log('opening comm port:  %s', portName);
var myPort = new SerialPort(portName, {
    baudRate: 57600,
});

function delay(millis) {
    var start = Date.now();
    var end = start + millis;
    while (Date.now() < end) { }
    console.log('slept: ' + (Date.now() - start));
}

function toggleLed() {
    if (!isHigh) {
        board.digitalWrite(ledPin, board.HIGH);
        isHigh = true;
        console.log('led on');
    } else {
        board.digitalWrite(ledPin, board.LOW);
        isHigh = false;
        console.log('led off');
    }
}

function blink(blinkDelay, blinkIterations)
{
    board.digitalWrite(ledPin, board.LOW);
    isHigh = false;
    for (var i = 0; i < (2*blinkIterations); i++) {
        toggleLed();
        delay(blinkDelay);
    }
    console.log('led is on: ' + isHigh);
}

function main() {
    toggleLed();
    var blinkDelay = 1000;
    var blinkIterations = 4;
    var loopInterval = (blinkDelay * blinkIterations * 2) + 1000;
    console.log('loop interval: ' + loopInterval);
    setInterval(function () {
        var start = Date.now();
        blink(blinkDelay, blinkIterations);
        console.log('main: ' + (Date.now() - start));
        console.log('loop interval: ' + loopInterval);
    });
}

console.log('creating board...');
board = new firmata.Board(myPort, function (err) {
  if (err) {
    console.log(err);
    board.reset();   
    return;       
  }
});

board.on('ready', function () {
    console.log('connected...');
    console.log('board.firmware: ', board.firmware);

    board.pinMode(ledPin, board.MODES.OUTPUT);
    main();
});
```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
root@Arduino:~/node# node blinkOnly
[2016-09-13 18:29:17.214] [INFO] console - opening comm port:  /dev/ttyATH0
[2016-09-13 18:29:17.320] [INFO] console - creating board...
[2016-09-13 18:29:22.544] [INFO] console - connected...
[2016-09-13 18:29:22.552] [INFO] console - board.firmware:  { name: 'StandardFirmataPlus.ino',
  version: { major: 2, minor: 5 } }
[2016-09-13 18:29:22.614] [INFO] console - led on
[2016-09-13 18:29:22.620] [INFO] console - loop interval: 9000
[2016-09-13 18:29:22.642] [INFO] console - led on
[2016-09-13 18:29:23.648] [INFO] console - slept: 1000
[2016-09-13 18:29:23.654] [INFO] console - led off
[2016-09-13 18:29:24.660] [INFO] console - slept: 1000
[2016-09-13 18:29:24.665] [INFO] console - led on
[2016-09-13 18:29:25.671] [INFO] console - slept: 1000
[2016-09-13 18:29:25.677] [INFO] console - led off
[2016-09-13 18:29:26.682] [INFO] console - slept: 1000
[2016-09-13 18:29:26.687] [INFO] console - led on
[2016-09-13 18:29:27.859] [INFO] console - slept: 1166
[2016-09-13 18:29:27.870] [INFO] console - led off
[2016-09-13 18:29:28.878] [INFO] console - slept: 1000
[2016-09-13 18:29:28.884] [INFO] console - led on
[2016-09-13 18:29:29.889] [INFO] console - slept: 1000
[2016-09-13 18:29:29.894] [INFO] console - led off
[2016-09-13 18:29:30.899] [INFO] console - slept: 1000
[2016-09-13 18:29:30.904] [INFO] console - led is on: false
[2016-09-13 18:29:30.909] [INFO] console - main: 8271
[2016-09-13 18:29:30.914] [INFO] console - loop interval: 9000
[2016-09-13 18:29:39.211] [INFO] console - led on
[2016-09-13 18:29:40.216] [INFO] console - slept: 1000
[2016-09-13 18:29:40.222] [INFO] console - led off
[2016-09-13 18:29:41.227] [INFO] console - slept: 1000
[2016-09-13 18:29:41.232] [INFO] console - led on
[2016-09-13 18:29:42.238] [INFO] console - slept: 1000
[2016-09-13 18:29:42.244] [INFO] console - led off
[2016-09-13 18:29:43.249] [INFO] console - slept: 1000
[2016-09-13 18:29:43.254] [INFO] console - led on
[2016-09-13 18:29:44.260] [INFO] console - slept: 1000
[2016-09-13 18:29:44.265] [INFO] console - led off
[2016-09-13 18:29:45.270] [INFO] console - slept: 1000
[2016-09-13 18:29:45.275] [INFO] console - led on
[2016-09-13 18:29:46.281] [INFO] console - slept: 1000
[2016-09-13 18:29:46.287] [INFO] console - led off
[2016-09-13 18:29:47.292] [INFO] console - slept: 1000
[2016-09-13 18:29:47.297] [INFO] console - led is on: false
[2016-09-13 18:29:47.303] [INFO] console - main: 8093
[2016-09-13 18:29:47.308] [INFO] console - loop interval: 9000

stardawg, 2016-09-13:
given

```js
function delay(millis) {
    var start = Date.now();
    var end = start + millis;
    while (Date.now() < end) { }
    console.log('slept: ' + (Date.now() - start));
}
```
seems to work fine...

```js
    setInterval(function () {
        toggle();
        delay(500);
    });
```
not so much....

```js
    while (true) {
        toggle();
        delay(500);
    }

````

rwaldron, 2016-09-13:
Thanks for the update. Based on the code above, it's clear that there is no issue with Firmata.js. JavaScript code running in Node.js will behave in unpredictable and unreliable ways when process blocking code is introduced, ie. that `delay(...)` function. By not specifying a time in ms for `setInterval`, the program expects that to execute when the current stack is exhausted + 0 ms, but exhausting the stack takes time as well, at best it will be sub millisecond, at worst it maybe 2-3ms. 

> should blink the led very slowly, 1s on 1s off

If you were using Johnny-Five, this would all be relatively simple: 

```js
var five = require("johhny-five");
var board = new five.Board({
  port: "/dev/ttyATH0"
});

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink(1000);
});
```

That operation is completely asynchronous.

closed 2016-09-13 by rwaldron