firmata/firmata.js #132: Is firmata compatible with NodeJS6.6 ARMv7?

jiapei100, 2016-09-16:
Hi, all:

```
> pi@raspberrypi:~ $ sudo npm install -g firmata
/usr/local/bin/firmata -> /usr/local/lib/node_modules/firmata/repl.js

> serialport@4.0.1 install /usr/local/lib/node_modules/firmata/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@6.6.0 (node-v48 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.6.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/firmata/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/local/lib/node_modules/firmata/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/6.6.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/local/lib/node_modules/firmata/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.13-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/local/lib/node_modules/firmata/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/local/lib/node_modules/firmata/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/local/lib/node_modules/firmata/node_modules/serialport
gyp ERR! node -v v6.6.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/firmata/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/firmata/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/firmata/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.13-v7+
node-pre-gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/firmata/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/local/lib/node_modules/firmata/node_modules/serialport
node-pre-gyp ERR! node -v v6.6.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_modules/firmata/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/firmata/node_modules/serialport/build/Release' (1)
/usr/local/lib
└── (empty)

npm ERR! Linux 4.4.13-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "firmata"
npm ERR! **node v6.6.0**
npm ERR! **npm  v3.10.3**
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
npm ERR! code 1
```

Can anybody give me a hand?

Cheers
Pei

reconbot, 2016-09-17:
You probably don't want to be using `sudo` or `-g` but since you are. Try;

`sudo npm install -g firmata --unsafe-perm`

jiapei100, 2016-09-17:
@reconbot 

> pi@raspberrypi:~ $ sudo npm install -g firmata --unsafe-perm
> /usr/local/bin/firmata -> /usr/local/lib/node_modules/firmata/repl.js
> 
> > serialport@4.0.1 install /usr/local/lib/node_modules/firmata/node_modules/serialport
> > node-pre-gyp install --fallback-to-build
> 
> node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v48-linux-arm.tar.gz 
> node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@6.6.0 (node-v48 ABI) (falling back to source compile with node-gyp) 
> make: Entering directory '/usr/local/lib/node_modules/firmata/node_modules/serialport/build'
>   CXX(target) Release/obj.target/serialport/src/serialport.o
> ../src/serialport.cpp: In function ‘int getIntFromObject(v8::Local<v8::Object>, std::string)’:
> ../src/serialport.cpp:90:52: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    return getValueFromObject(options, key)->ToInt32()->Int32Value();
>                                                     ^
> ../src/serialport.cpp: In function ‘double getDoubleFromObject(v8::Local<v8::Object>, std::string)’:
> ../src/serialport.cpp:102:53: warning: ‘v8::Local<v8::Number> v8::Value::ToNumber() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8166): Use maybe version [-Wdeprecated-declarations]
>    return getValueFromObject(options, key)->ToNumber()->NumberValue();
>                                                      ^
> ../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
> ../src/serialport.cpp:175:31: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>      int fd = argv[1]->ToInt32()->Int32Value();
>                                ^
> ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
> ../src/serialport.cpp:195:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    int fd = info[0]->ToInt32()->Int32Value();
>                              ^
> ../src/serialport.cpp:220:116: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    baton->baudRate = Nan::Get(options, Nan::New<v8::String>("baudRate").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
>                                                                                                                     ^
> ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Write(Nan::NAN_METHOD_ARGS_TYPE)’:
> ../src/serialport.cpp:256:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    int fd = info[0]->ToInt32()->Int32Value();
>                              ^
> ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)’:
> ../src/serialport.cpp:376:32: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    baton->fd = info[0]->ToInt32()->Int32Value();
>                                 ^
> ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)’:
> ../src/serialport.cpp:490:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    int fd = info[0]->ToInt32()->Int32Value();
>                              ^
> ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)’:
> ../src/serialport.cpp:538:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    int fd = info[0]->ToInt32()->Int32Value();
>                              ^
> ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Drain(Nan::NAN_METHOD_ARGS_TYPE)’:
> ../src/serialport.cpp:596:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    int fd = info[0]->ToInt32()->Int32Value();
>                              ^
>   CXX(target) Release/obj.target/serialport/src/serialport_unix.o
> ../src/serialport_unix.cpp: In function ‘OpenBatonPlatformOptions* ParsePlatformOptions(const v8::Local<v8::Object>&)’:
> ../src/serialport_unix.cpp:44:109: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.6.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
>    result->vmin = Nan::Get(options, Nan::New<v8::String>("vmin").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
>                                                                                                              ^
[... cut ...]
>     │   │ ├── inherits@2.0.1 
>     │   │ ├── isarray@1.0.0 
>     │   │ ├── process-nextick-args@1.0.7 
>     │   │ ├── string_decoder@0.10.31 
>     │   │ └── util-deprecate@1.0.2 
>     │   └── uid-number@0.0.6 
>     └─┬ object.assign@4.0.4 
>       ├─┬ define-properties@1.1.2 
>       │ └── foreach@2.0.5 
>       ├── function-bind@1.1.0 
>       └── object-keys@1.0.11 
> 
> pi@raspberrypi:~ $ 
> 

I think it seems to be OK now??? Is it fully functional??? Let me try...

Thank you reconbot...

soundanalogous, 2017-07-08:
Closing since it looks like this issue was resolved.

closed 2017-07-08 by soundanalogous