firmata/firmata.js #133: Close board and connection to Arduino

troywweber7, 2016-09-27:
I've been browsing the firmata.js docs and code and I cannot seem to find a function that disconnects from the Arduino. For example

```javascript
let board = new firmata.Board('/dev/arduino' function(err){};);
```

starts up the connection to the Arduino board, but how do you close that connection? Do you simply `delete board`? Or is there some `firmata.disconnect(board)` function I'm missing?

troywweber7, 2016-09-27:
The closest I found was `board.reset()` but this does not appear to disconnect from the serial device altogether.

ericogr, 2016-09-28:
Maybe add an exit function:

```
Board.prototype.exit = function(callback) {
  this.transport.close(callback);
};
```

?

ericogr, 2016-10-03:
#135

troywweber7, 2016-10-03:
Sorry I never responded on here. That worked for me.

troywweber7, 2016-10-03:
marking as closed since #135 is open.

closed 2016-10-03 by troywweber7