firmata/firmata.js #137: Promise of completion

troywweber7, 2016-10-21:
I have poured over the `firmata.js` file extensively. For some of the functions available on a board there is a callback that ensures the data has been sent because a response comes in. For other functions, you just send it and hope it happens. What is the preferred way to ensure a function like `board.setSamplingInterval` has actually completed? There is no callback function or response from the Arduino Board associated with this function.

My thought for how to resolve this: maybe I should call a dummy `queryPinState` function right after and use its callback? Or maybe using `board.transport.drain` will ensure that whatever I just queued up will be sent before I proceed?

soundanalogous, 2016-10-21:
There is no way to do this because StandardFirmata running on the Arduino does not implement an RPC protocol. Many commands are only one direction.

soundanalogous, 2016-10-21:
Also there is a proposal to add the ability to query the sampling interval: https://github.com/firmata/protocol/pull/34.