firmata/firmata.js #140: stepperStep peculiar behavior with acceleration/deceleration

troywweber7, 2016-11-09:
I'm not certain whether the problem is on the `firmata.js` side or the `ConfigurableFirmata` side. I opened an issue over at `ConfigurableFirmata` because I suspect the issue is in `FirmataStepper.cpp`: https://github.com/firmata/ConfigurableFirmata/issues/59.

But I think it would be helpful to document the odd behavior I'm seeing when issuing `stepperStep()` commands in REPL:

- `board.stepperStep(0, 1, 200, 500, `**`0, 0,`**` ()=>console.log('DONE'))` steps properly and logs "DONE"
- `board.stepperStep(0, 1, 200, 500, `**`50, 0,`**` ()=>console.log('DONE'))` steps properly and logs "DONE", **but no acceleration is actually applied**
- `board.stepperStep(0, 1, 200, 500, `**`0, 50,`**` ()=>console.log('DONE'))` steps properly and logs "DONE", **but no deceleration is actually applied**
- `board.stepperStep(0, 1, 200, 500, `**`50, 50,`**` ()=>console.log('DONE'))` appears to accelerate and decelerate, **but never prints "DONE"**

Any help anyone has on how to resolve this behavior and cause it to accelerate and decelerate properly would be greatly appreciated.

soundanalogous, 2016-11-09:
This is partly a documentation error. You must supply both acceleration and deceleration values, not one or the other. If one of them is zero, then no acceleration or deceleration will be applied. I forget the exact reason why I had to implement it [this way](https://github.com/firmata/ConfigurableFirmata/blob/master/src/utility/FirmataStepper.cpp#L159-L171) since It's been a few years since writing it.

Why the following is not printing "DONE" is a bug:
```javascript
board.stepperStep(0, 1, 200, 500,50, 50,()=>console.log('DONE'));
```

I'm not sure if it's on the JS side or the firmware side yet. Will try to look into it tonight if I get home from work at a reasonable time.

soundanalogous, 2016-12-04:
This callback issue is now fixed in ConfigurableFirmata v2.9.2: https://github.com/firmata/ConfigurableFirmata/releases/tag/2.9.2

You must still specify both and acceleration value and a deceleration value, or neither. The ability support only acceleration or only deceleration is not supported. I'll file a separate issue to update the documentation in firmata.js.

closed 2016-12-04 by soundanalogous