firmata/firmata.js #141: is serialConfig required before using serialWrite/Read/etc?

jywarren, 2016-11-27:
I'd like to submit a PR to clarify this if so: https://github.com/firmata/firmata.js#serial

Also, I can add a clarification if the `board.serialConfig(options)` "options" parameter should have property names like:

```js
{
  portId: <value>,
  baud: <value>
}
```

I think it's a little ambiguous as it's shown currently, and it'd be preferable to have true parse-able JSON shown in the README.

Thanks a lot! Love this library!

soundanalogous, 2016-11-27:
Yes in Firmata you must always first send a config command for almost every type of communication.

soundanalogous, 2016-11-27:
`portId` is required, `baud` is optional (defaults to 57600).

In hindsight, the signature of serialConfig should have probably been:
`Board.prototype.serialConfig = function(portId, options)`

jywarren, 2016-11-27:
Well, next major release, perhaps. OK - opened a PR!

commit e12c065c (2016-11-28): Merge pull request #142 from jywarren/serial-docs

proposed changes to serial config and usage docs, closes #141
closed 2016-11-28 by soundanalogous