firmata/firmata.js #146: Speed/Acceleration/Deceleration Units

troywweber7, 2016-12-20:
This is a minor issue, but I just wanted to point out that the `stepperStep` command expects speed in .01 rad/s (or accel/decel in .01 rad/s^2). It works, but this is sort of odd and it should be pointed out that if you use micro-stepping, this will alter the speed units. (i.e. if you do half-stepping, it will take twice as many steps to move the same speed). I'm mainly pointing this out for anyone interested.

dtex, 2016-12-20:
@troywweber7 A [new protocol for stepper](https://github.com/firmata/protocol/blob/master/stepper-2.0.md) has been proposed. Everything is now passed in steps/second or steps/second^2. Mapping to/from radians, degrees, revolutions or any other units will be handled by the client.

troywweber7, 2016-12-21:
I saw the new protocol, I'll try to take a look at the new stuff and let you know my thoughts.

In the meantime, for those still using 1.0, I think I answered my own question: each byte is only transferring 7 bits of data (which is a bummer, but I'm sure there is a reason for it), so speed and accel values max out at 2^14-1 for version 1.0. Correct me if I'm wrong.

By the way, if you don't catch this maxing out on the javascript side, there will be overflow issues (I'm dealing with such overflow issues right now). Same goes for the number of steps, I believe.

soundanalogous, 2018-01-05:
The new and improved accelstepper feature has landed.

closed 2018-01-05 by soundanalogous