firmata/firmata.js #148: example of using servo?

zoooofu, 2017-01-09:
I have known firmata.js providing **servoWrite(pin, degree)** and **servoConfig(pin, min, max)** to control a servo, can anyone give an example of using **servoConfig** api? On the other hand, why there is no any api of reading servo's degree, like servo.read() function provided in Arduino? Thanks.

soundanalogous, 2017-01-09:
The is an example in the examples directory: https://github.com/firmata/firmata.js/blob/master/examples/servo-config.js.

Regarding servo.read(), I'm soliciting feature requests for a new Firmata Servo 2.0 api here: https://github.com/firmata/protocol/issues/78. So please add a request there.

zoooofu, 2017-01-12:
@soundanalogous Thanks for your comment. I have commented something of Servo 2.0 api under your request, hopefully it would make sense.

rwaldron, 2017-01-13:
servo.read() doesn't actually read the servo hardware, all it does is return the present position computed in terms of the latest tick in microseconds. The present position is always whatever your program last told it to be, so your code always knows exactly what it last wrote to the servo.

zoooofu, 2017-01-13:
@rwaldron Thanks for pointing out my misunderstand of servo.read(). Here is the description of it from [Arduino official site](https://www.arduino.cc/en/Reference/ServoRead).

soundanalogous, 2017-01-13:
There is no guarantee the value returned by `read()` is correct. The servo could have been obstructed or some other reason that caused it not to move the full amount expected per the last call to `write()`. However returning the `read()` value from the firmware (the Servo library) would be more accurate than relying on the last write() value from the Firmata client, but that also comes with the cost of increased latency.

rwaldron, 2017-01-16:
> Here is the description of it from Arduino official site.

My summary was based on a review of Servo library source, to determine how it implemented `read()`

closed 2017-03-22 by rwaldron