firmata/firmata.js #150: Modernizing the codebase

rwaldron, 2017-01-13:
I've had some work in progress sitting in a stash for a very long time. I tried it out tonight and of course it needed tons of updating. When I got to changing relevant portions of code from function declarations (constructors) to class definitions, mocha exploded: 

```js
Running "mochaTest:files" (mochaTest) task
>> Mocha exploded!
>> SyntaxError: Unexpected reserved word
```

Which is the last effing straw for mocha. I'm going to start rewriting all the tests to use nodeunit.

rwaldron, 2017-01-13:
On further investigation, it was actually this: 

```diff
-module.exports = com;
+export default com;
```

rwaldron, 2017-01-16:
I actually didn't mean to close this.

rwaldron, 2017-01-16:
This is on hold until I can do a study of all projects that depend on Firmata.js and which also stub the constructor with sinon.js

Ahmed-90, 2017-11-30:
A friendly suggestion is that instead of worrying about all the other projects and how they depends on this module, maybe start clean with firmata2.js or something like that, with all the fancy async/await or maybe even write it in typescript or at least provide native typescript support to make testing easier etc..?

Others can upgrade later whenever they feel like it?

rwaldron, 2019-01-15:
This has been completed in 2.0.0

closed 2019-01-15 by rwaldron