firmata/firmata.js #151: Unable to read bytes from I2C arduino

miguel719, 2017-02-10:
I'm trying to read bytes from an slave arduino. But i'm unable to get the correct bytes values.

The slave arduino code:
```
#include <Wire.h> 

void setup() {
  Wire.begin(8);
  Wire.onRequest(requestEvent);
}

void loop() {
  delay(500); 
}

void requestEvent() {
  uint8_t buffer[4];
  buffer[0] = 12;
  buffer[1] = 23;
  buffer[2] = 39;
  buffer[3] = 78;
  Wire.write(buffer, 4);
}
```
The Firmata code:
```

let firmata = require("firmata");

board = new firmata('COM3', {samplingInterval: 1000});
board.on("ready", function() {
    this.i2cConfig();
    this.i2cRead(8,4, function(data) {
      console.log("received data");
      console.log(data);
    });
});

```
The response i get is:
```
received data
[189,255,255,255]
```

soundanalogous, 2017-02-10:
I'm assuming you have an Arduino running StandardFirmata (or one of the variants) connected to a computer running firmata.js and then a second Arduino acting as an I2C slave connected the Arduino running StandardFirmata, right?

The first thing I suspect is that 500ms delay is messing things up so remove that from your Arduino code since it's blocking and unnecessary.

rwaldron, 2017-03-22:
Using exactly the arduino slave code you provided, I created the following program: 

```js
var Board = require("firmata");

Board.requestPort(function(error, port) {
  if (error) {
    console.log(error);
    return;
  }

  var board = new Board(port.comName); // no change to sampling interval

  board.on("ready", function() {
    this.i2cConfig();
    this.i2cRead(8,4, function(data) {
      console.log("received data");
      console.log(data);
    });
  });
});
```

Which produces the following results: 

```
$ node examples/i2c-slave-request.js
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
```

When I added ` {samplingInterval: 1000}`, the results are the same: 

```
$ node examples/i2c-slave-request.js
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
received data
[ 12, 23, 39, 78 ]
```