firmata/firmata.js #152: pwmWrite doesn't allow values > 14 bits on pin numbers < 16

soundanalogous, 2017-02-26:
For future proofing, we should allow pwm/analog writes of values > 14 bits on pins 0 - 15. Currently higher res writes are only available for pins > 15:

https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L776-L806

rwaldron, 2017-03-22:
I'm certainly all for future proofing—do you have an idea of how we might accomplish this?

rwaldron, 2017-03-22:
I wonder if we could just switch to using `EXTENDED_ANALOG` for all pins? (I haven't actually looked to see if that's doable)

soundanalogous, 2017-03-22:
Switching to `EXTENDED_ANALOG` is one option. Another is changing line 781 to:

```
if (pin > 15 || value >= 0x00040000 - 1) {
```

soundanalogous, 2017-03-22:
correction (no need for the -1)
```
if (pin > 15 || value >= 0x00040000) {
```