firmata/firmata.js #161: Firmata over wifi

gokhandilek, 2017-04-04:
I am planning to get this library working with wifi.
Do you see any issues with that?
Plannig to get the mkr1000 to talk to a remote cloud server. Do I need to be on the same network as the mkr1000? Or can I connect a remote server to a remote mkr1000 over firmata protocol?
Thanks.

soundanalogous, 2017-04-04:
firmata.js already works with WiFi although it's not well documented. See the blink-firmata-client.js and blink-firmata-server.js examples here: https://github.com/soundanalogous/firmata-client-examples/tree/master/firmata-js/network.

Then read through the notes in [wifiConfig.h](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/wifiConfig.h) to see how to configure the MKR1000 as a TCP client or server. It's configured as a TCP server by default and then use blink-firmata-client.js as an example. If you'd rather configure the MKR1000 as a TCP client, uncomment line 114 in wifiConfig.h and use blink-firmata-server.js as an example.

soundanalogous, 2017-04-04:
If you have a server on a static IP you can use that as the TCP server and then the MKR1000 does not need to be on the same network, however you will have much higher latency than if the board and server are on the same local network.

gokhandilek, 2017-04-04:
Thanks for your reply.
I am trying to avoid a local server if possible. 
But if the latency is going to be a problem, then I'd rather have the arduino uno directly connected via USB(which is what I have at the moment).
I don't see any other advantages of having the firmata over wifi. You can of course avoid the data cable which can make the system a bit more tidy.
Other than latency issues, mkr1000 does not provide much power to drive other devices like for example..
Do you think mkr1000 can drive this stepper driver?
http://www.leadshine.com/UploadFile/Down/EMhm_V1.1.pdf
It seems that the stepper driver expects 3.5V as digital input.

closed 2017-04-07 by gokhandilek