firmata/firmata.js #162: Connecting to different boards within the same server

gokhandilek, 2017-04-04:
I have a node js app which I can connect the arduino board to.
Is it possible to connect to different boards within the same server?
Especially, if I have two boards on two locations, can I connect the node js app to these two boards without any issue?

troywweber7, 2017-04-04:
"without any issue" is a relative phrase. Short answer, it can be done.

Long answer: assuming your node.js server is running on hardware with USB ports, you can connect to as many firmata-supported boards as you like. And if you look hard enough, you'll find there are wireless versions of firmata as well (both firmware and JavaScript-side), though I have not (yet) personally tested said libraries.

gokhandilek, 2017-04-04:
You can define the device something like dev/usb099203 if you are using USB but how does the device identification using Wifi work?

gokhandilek, 2017-04-04:
I am guessing if you define the wifi port per device, it should be enough?

rwaldron, 2017-04-04:
If you know the system path or port name of the specific devices, you can use them to explicitly create `Board` instances with that data: 

```js
var Board = require("firmata");

Promise.all([
  new Promise(resolve => {
    var a = new Board("/dev/cu.modemAAAA", error => {
      if (!error) {
        resolve(a);
      }
    });
  }), 
  new Promise(resolve => {
    var b = new Board("/dev/cu.modemBBBB", error => {
      if (!error) {
        resolve(b);
      }
    });
  }), 
]).then(boards => {

  console.log("Both boards are connected and ready to go");

});
```

I'm mobile so this code is untested at the moment.

closed 2017-04-08 by gokhandilek