firmata/firmata.js #163: Security and authentication

gokhandilek, 2017-04-04:
How can one secure the firmata  protocol between the client and the server?
2 scenerios:

I have a device controlled by an arduino uno which acts as a thin client & connected to a host computer via USB. The host has a REST API and front end which has buttons and controls for various different tasks.

The same device can be controlled by arduino mkr1000 acts as a thin client and connected to a cloud server.

The first scenerio is pretty straight forward - as long as noone can connect to the arduino physically - i am in good shape. Of course I need to secure the REST API and authentication.

The second scenerio is a little scary.
Can the firmata protocol manipulatad(man in the middle attack before it reaches the cloud server)?
If the cloud server is directly communicating to the device using the firmata protocol, where is the encryption in this? 
if I register arduino mkr1000 as thin client & specify the cloud host computer, can a bad guy create a host and somehow control my device?

soundanalogous, 2017-04-07:
I'm not sure. I'm not a security expert. I'd need someone who is to chime in here. If you need a secure connection, Firmata WiFi may not be a good option for you, at least not for the near future. You would be better off making a USB/Serial connection to the board and accessing that from a secure server application.

gokhandilek, 2017-04-07:
What are the chances of someone manipulating the protocol or interpreting the protocol?

Can I use VPN tunneling?

Arduino Yun Client - > Firmata > VPN Tunnel < Firmata < Remote Server

closed 2017-04-16 by gokhandilek