firmata/firmata.js #171: Unable to install the package

kevinmamaqi, 2017-06-14:
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/Cellar/node/7.0.0/bin/node',
1 verbose cli   '/usr/local/bin/npm',
1 verbose cli   'install',
1 verbose cli   'firmata' ]
2 info using npm@3.10.8
3 info using node@v7.0.0
4 silly loadCurrentTree Starting
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 silly rollbackFailedOptional Starting
8 silly rollbackFailedOptional Finishing
9 silly runTopLevelLifecycles Finishing
10 silly install printInstalled
11 verbose stack Error: Failed to parse json
11 verbose stack Trailing comma in object at 27:3
11 verbose stack   }
11 verbose stack   ^
11 verbose stack     at parseError (/usr/local/lib/node_modules/npm/node_modules/read-package-json/read-json.js:390:11)
11 verbose stack     at parseJson (/usr/local/lib/node_modules/npm/node_modules/read-package-json/read-json.js:79:23)
11 verbose stack     at /usr/local/lib/node_modules/npm/node_modules/read-package-json/read-json.js:48:5
11 verbose stack     at /usr/local/lib/node_modules/npm/node_modules/graceful-fs/graceful-fs.js:78:16
11 verbose stack     at tryToString (fs.js:425:3)
11 verbose stack     at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:412:12)
12 verbose cwd /Users/kevinmamaqi/Desktop/SST
13 error Darwin 16.6.0
14 error argv "/usr/local/Cellar/node/7.0.0/bin/node" "/usr/local/bin/npm" "install" "firmata"
15 error node v7.0.0
16 error npm  v3.10.8
17 error file /Users/kevinmamaqi/Desktop/SST/package.json
18 error code EJSONPARSE
19 error Failed to parse json
19 error Trailing comma in object at 27:3
19 error   }
19 error   ^
20 error File: /Users/kevinmamaqi/Desktop/SST/package.json
21 error Failed to parse package.json data.
21 error package.json must be actual JSON, not just JavaScript.
21 error
21 error This is not a bug in npm.
21 error Tell the package author to fix their package.json file. JSON.parse
22 verbose exit [ 1, true ]

rwaldron, 2017-06-14:
Please check your file `/Users/kevinmamaqi/Desktop/SST/package.json`  in https://jsonlint.com/

closed 2017-06-14 by rwaldron

kevinmamaqi, 2017-06-14:
Solved.