firmata/firmata.js #172: Firmata.js altogether with browserify don't recognize the Serial Port:

kevinmamaqi, 2017-06-14:
On my mac I connect arduino and run:
ls /dev/tty.* // Find Serial Port number and name.
/dev/tty.Bluetooth-Incoming-Port **/dev/tty.usbmodem1421**

Then on my main.js (before bundling with browserify) I have the following:

```
var Board = require("firmata");
var board = new Board("/dev/tty.usbmodem1421)");

  board.on("open", function() {
    console.log("  ✔ open");
  });

/*board.on("ready", function() {
  	// Arduino is ready to communicate
	this.analogRead(0, function(value) {
	  console.log("The value of pin A0 is " + value + " as reported at the sampling interval");
	});
});*/
```

Get the following message: _**Uncaught Error: No access to serial ports. Try loading as a Chrome Application.**_

rwaldron, 2017-06-23:
Are you trying to run this in a non-Electron, non-NW.js browser?

kevinmamaqi, 2017-06-23:
I think so, I was using chrome. Not aware of what non-NW.js browsers are.

morganrallen, 2018-02-14:
@kevinmamaqi Chrome doesn't have access to the filesystem, device descriptor or USB port you're trying to use. Browserify cannot help you with this as it's a browser security limitation causing this problem.

closed 2018-02-19 by kevinmamaqi