firmata/firmata.js #180: Question re: servoWrite

dtex, 2017-09-10:
in firmata.js:
````
Board.prototype.servoWrite = function(pin, value) {
  // Values less than 544 will be treated as angles in degrees
  // (valid values in microseconds are handled as microseconds)
  this.analogWrite.apply(this, arguments);
};
````
I've followed the code path but cannot find where values < 544 are treated as angles. Can you give me a push in the right direction? I'm working on adding support for multi-turn servos and want to make sure I understand what's happening in the backend.

soundanalogous, 2017-09-10:
I don't think that comment is accurate unless it is assuming some custom Firmata running on the board (not StandardFirmata). On the firmware side, an analog write message received for a Servo pin will simply [call the write method on the Servo object](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L386). There is no logic in StandardFirmata or ConfigurableFirmata to call [writeMicroseconds](https://www.arduino.cc/en/Reference/ServoWriteMicroseconds) if the value is > 544.

soundanalogous, 2017-09-10:
Maybe a legacy version of Servo.write worked like this (and at some point Arduino split out the time-based write to a new writeMicroseconds method). According to the latest Servo library documentation however, [Servo.write](https://www.arduino.cc/en/Reference/ServoWrite) is only for setting the angle.

soundanalogous, 2017-09-10:
Did a little digging (I should have done this first) and actually Servo.write does work as described in the comment in firmata.js. It's just not documented anywhere other than in the Servo library code: https://github.com/arduino-libraries/Servo/blob/master/src/avr/Servo.cpp#L264-L273. According to the code, values < 544 are capped to the 0 to 180 range, so you can't set an angle of 220 degrees for example, that would still just result in 180 degrees.

dtex, 2017-09-10:
Excellent!

Now I need to convince all the io plugin authors to match that. We are sacrificing a ton of possible servo resolution in Johnny-Five.

Thank you for digging into that.

closed 2017-09-10 by dtex