firmata/firmata.js #185: board.reportDigitalPin(digitalPinNumber, 0) has no effect

pichlermi, 2018-03-02:
From the documentation:

>> To stop reporting digital values for a pin, call board.reportDigitalPin(digitalPinNumber, 0). To restart, call digitalRead(pin,callback) or use board.reportDigitalPin(digitalPinNumber, 1) if you don't want to call digitalRead again. <<

Reproduce the error:

board.pinMode( 2, board.MODES.INPUT );
board.digitalRead(2, function(value) {
  console.log("The value of digital pin 2 changed to: " + value);
});
board.reportDigitalPin(2, 0);

The log message is still written to the console, when Pin 2 is toggled.

Hack:

board.pinMode( 2, board.MODES.INPUT );
cb = function(value) {
  console.log("The value of digital pin 2 changed to: " + value);
}
board.digitalRead(2, cb);
board.removeListener( "digital-read-2", cb )

Now, no message is written.

lilliesAndRoses, 2018-04-05:
Are you trying to work around a scenario where you want to read the values infrequently and are using analogRead for that?

lilliesAndRoses, 2018-04-05:
Are you trying to read the value of a pin infrequently using analogRead or digitalRead, and do not want the callbacks there after? 

I was trying to do that and made my code look as follows

```javascript
function readPin(pinId) {
    board.pinMode( pinId, board.MODES.INPUT );
    board.digitalRead(pinId, function(value) {
       console.log("The value of digital pin changed to: " + value);
       board.reportDigitalPin(pinId, 0);
 });
}
```
With this change, I got warning of maxEventListeners after calling the function _readPin_ 10 times. 

I think there is a need of _analogReadOnce_ and _digitalReadOnce_ type of functions. The code there can simply call "_this.once_" instead of "_this.addListener_" so that the above code would work and user would not need to tamper with the listeners directly.

```javascript
Board.prototype.analogReadOnce = function(pin, callback) {
  this.reportAnalogPin(pin, 1);
  this.once("analog-read-" + pin, callback);
};
```

lilliesAndRoses, 2018-04-06:
@rwaldron Thanks. The commits that you made also help solve the problem. The only change I made to my code is moved the reportDigitalPin(pinId, 0) as the first thing to do in the callback.
```javascript
function readPin(pinId) {
    board.pinMode( pinId, board.MODES.INPUT );
    board.digitalRead(pinId, function(value) {
       board.reportDigitalPin(pinId, 0);
       console.log("The value of digital pin changed to: " + value);
 });
}
```