firmata/firmata.js #186: Controlling Tinkerkit's Braccio with firmata.js

cnizo, 2018-04-06:
Hello!
I am sorry if this is an obvious question, but I have been trying to control Braccio from Tinkerkit with firmata.js, but the servoWrite function has no effect on it. 
I have uploaded StandardFirmata on Arduino UNO. From the Braccio library for Arduino I saw that the servos are attached to pins 11, 10, 9, 6, 5 and 3, so can't I just use servoWrite with this pins and the desired angles?

Thanks!

dtex, 2018-04-08:
It certainly looks like it should work. Could you share your code? It might make it easier to diagnose.

cnizo, 2018-04-09:
This is what I tried to do. I dig a little and when I run the code the function "Board.prototype.pwmWrite" on firmata.js is called. 

`var Board = require('firmata');

    Board.requestPort(function (error, port) {
      if (error) {
        console.log(error);
        return;
      }
    
      board = new Board(port.comName, { samplingInterval: 1000 });
    
      board.on('open', function () {
        console.log('  board opened');
        alert('board opened')
      });
    
      board.on('ready', function () {
        console.log('  board ready');
        board.pinMode(11,board.MODES.SERVO);
        board.pinMode(10,board.MODES.SERVO);
        board.pinMode(9, board.MODES.SERVO);
        board.pinMode(6, board.MODES.SERVO);
        board.pinMode(5, board.MODES.SERVO);
        board.pinMode(3, board.MODES.SERVO);

        board.servoWrite(11,85);
        board.servoWrite(10,85);
        board.servoWrite(9,85);
        board.servoWrite(6,85);
        board.servoWrite(5,85);
        board.servoWrite(3,60);
      });
    });`