firmata/firmata.js #187: Can't communicate over UART.

overflowz, 2018-04-06:
Hey everyone,

Found many likely issues here of this but still was not unable to make it work.

I'm trying to connect Arduino and Raspberry Pi 3 over UART. Tried many ways but all of them failed.

What I have now:

ttyAMA0 port on raspberry -- which works pretty fine. Wrote a custom sketch that echoes back what you sent, tested and worked fine like a charm.

Whenever I'm writing StandardFirmataPlus and using either Johnny-Five or firmata's repl.js, it never gets connected and just hanging.

Any thoughts?

rwaldron, 2018-04-06:
@soundanalogous any thoughts?

soundanalogous, 2018-04-06:
Is the baud rate set to 57600 on the Pi 3?

overflowz, 2018-04-06:
@soundanalogous if I remember correctly -- yes. I started serial port on arduino side on baud rate of 57600 and messages were receiving properly with `echo "test" > /dev/ttyAMA0` with echo sketch.

soundanalogous, 2018-04-06:
Hummm since you're using StandardFirmataPlus I assume you may be trying to use the Firmata Serial feature to communicate with the Pi 3. If that is the case, then you can't use RX0/TX0 since they are used for the primary communication with the board. You'd need to connect RX1/TX1 or higher. That serial data is then routed through RX0/TX0 to your computer (if you're the Due USB connection).

overflowz, 2018-04-06:
Hmm, I'll post what I used:
StandardFirmataPlus with:
1) #define FIRMATA_SERIAL_FEATURE
2)
````
Serial1.begin(57600);
Firmata.begin(Serial1);
````
and also used Arduino Due's RX1/TX1 pins rather than RX0/TX0
3) both of them.

USB was completely removed when testing these and still no luck. Am I missing something?

soundanalogous, 2018-04-06:
#1 and #2 above are for two separate things.

FIRMATA_SERIAL_FEATURE is the ability to use a UART through Firmata. This is different than using a UART as the Transport for Firmata. The Serial Feature is for communicating with things like a GPS receiver that requires a UART connection, or for communicating with another board using a protocol of your own definition.

I'm not sure if you're trying to create a custom Serial protocol to communicate between the Due and the Pi or if you're trying to use Firmata to communicate between the two (where you write a firmata.js script that runs on the Pi to communicate with the Due). If it's the latter, then you don't need StandardFirmataPlus. That still doesn't resolve your issue, but it will help me understand exactly what your setup is and what you're trying to accomplish here.

overflowz, 2018-04-06:
as for FIRMATA_SERIAL_FEATURE, I was just playing with it -- was not sure what it was for.

As for the setup, I'm literally trying to communicate rpi to arduino due using firmata with UART (RX/TX pins) rather than USB one.

soundanalogous, 2018-04-06:
TLDR: I think the issue may actually be that there is no auto-reset on the Board when you wire directly to the UART.

Historically Arduino boards reset upon establishing a serial connection. This was done by wiring the DTR pin in the UART connection. This meant any Sketch on the board would restart upon establishing a serial connection. For backwards compatibility when boards started using a USB serial, they kept this convention (I'm not sure how it's wired up exactly to achieve this). Firmata adapted this behavior early on (back when there were only a couple of Arduino boards) by sending the Firmata version number when the Sketch starts. Firmata client libraries then adapted this a start signal. If this signal is missing, the client library won't work (very fragile today, but it was common for many years). This really needs to be re-architected so that the client does not need to respond to a start signal from the board.

To test if this is actually the issue, you can try to run the script on your Pi, then immediately hit the hardware reset on the Due. If that works, then indeed this is the issue. It may take a few tries to get the timing of when to press the HW button just right. You may need to extend the firmata.js timeout as well.

overflowz, 2018-04-07:
Well, still no luck. Tried resetting as well, many, many times.

Here's the screenshot of debug information if it helps somehow, after entering port number & clicking reset one time.

[image]

No matter how many times I click reset or trying to do synchronously, result is always the same -- it hangs (johnny-five throws error to check if I'm using correct version of firmata).

Also tried to debug, to make sure that data is coming for sure and it does. Firmata reports version as it should, like it did on Serial Monitor, even if I do `cat /dev/ttyAMA0` but for some reason, it does not work with firmata itself.

am I missing something?

soundanalogous, 2018-04-07:
You're getting the right sequence on startup. The protocol version message followed by the firmware name/version message. For some reason firmata.js is not responding. Can you paste in your full JavaScript code?

overflowz, 2018-04-07:
I'm using firmata's repl.js file located in `node_modules/firmata` folder to test it out.

One more question is, why I'm getting huge data over Serial and only two bytes from other Serial ports(1, 2, 3)?

soundanalogous, 2018-04-07:
Have you tried examples/blink.js?

overflowz, 2018-04-07:
I did, I'm getting same errors.

soundanalogous, 2018-04-07:
I have a Pi 3 and a Due, I'll have to replicate your setup to get a better understanding of what the issue is. Won't be able to get to that until tomorrow at the earliest.

overflowz, 2018-04-07:
Well, sure! I'll write down what I'm using right now:

1) sketch: StandardFirmata / StandardFirmataPlus.
Also modified one, replacing following lines (https://github.com/firmata/arduino/blob/master/examples/StandardFirmataPlus/StandardFirmataPlus.ino#L801-L802) with this:
````
Serial3.begin(57600);
Firmata.begin(Serial3);
````

2) Wire configuration:
Arduino RX3/TX3 (or RX/TX in case of original sketch) -> Rpi TX/RX (GPIO 14 and 15)
Arduino GND -> Rpi GND
Arduino 5V -> Rpi 5V (or with usb cable, tried both).

3) Rpi configuration:
I have disabled console & bluetooth use of UARTs. (followed this guide https://www.raspberrypi.org/documentation/configuration/uart.md)

Double checked on this and communication works as it should, at least echo sketch is working pretty well.

Raspberry is connected to my PC via USB to power it up and also either it powers up arduino or connecting with USB on my PC too, tried them both.

If you'll need more information or I'm missing something, let me know. I'm also not an expert of wiring, but friend of mine is helping and he assures me that everything should be working fine.

Best regards.

soundanalogous, 2018-04-07:
You should be able to use `Serial` rather than Serial3/2/1. I was unsure initially which method of Serial communication you were referring to.

Here's one more thing to try. This will skip the capability query that normally replies with a 500+ byte packet that may overflow your UART and cause issues.

```javascript
var Board = require("../");

Board.requestPort(function(error, port) {
  if (error) {
    console.log(error);
    return;
  }

  var board = new Board(port.comName, {
    skipCapabilities: true,
    pinCount: 66,
    analogPins: [54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65]
  });

  board.on("ready", function() {
    var pin = 13;
    var state = 1;

    board.pinMode(pin, board.MODES.OUTPUT);

    setInterval(function() {
      board.digitalWrite(pin, (state ^= 1));
    }, 500);
  });
});
```

overflowz, 2018-04-07:
I'm getting `No Acceptable Port Found`

soundanalogous, 2018-04-07:
Hummm, sounds like port detection is not working. You can remove the outer Board.requestPort wrapper function and replace `port.comName` with a hard coded string.

soundanalogous, 2018-04-07:
Also `var Board = require("firmata");` rather than `var Board = require("../");`

overflowz, 2018-04-07:
It just hangs like this:

[image]

as for `require('firmata');` no worries on that.

soundanalogous, 2018-04-07:
And you're using the PL011 rather than mini UART on the Pi right?

overflowz, 2018-04-07:
Yup. For some reason, the code you gave me, it triggers 'ready' event when clicking reset button, but LED is not toggling and as I see on the DEBUG screen, it is reading and writing data often (which was not happening before).

soundanalogous, 2018-04-07:
Hummm so it looks like transmission is working from Arduino to Pi (you're seeing the firmware name and version being logged), but not from Pi to Arduino (LED does not blink).

soundanalogous, 2018-04-07:
Can you log what bytes are being sent from the Pi to the Arduino?

overflowz, 2018-04-07:
Not sure how to do that, but I'll try.

soundanalogous, 2018-04-07:
In firmata.js, insert a log statement at line 798, logging `data`

soundanalogous, 2018-04-07:
Actually that should be correct, I'd actually need logging on the Arduino side. I'll have to send you a modified version of StandardFirmata.

overflowz, 2018-04-07:
Would appreciate it.

soundanalogous, 2018-04-07:
Compile and upload this Sketch to your Due: https://gist.github.com/soundanalogous/e10be1c5c0e4031e5aba0769ce5bc1e8

Then run this test script:
```javascript
var Board = require("firmata");

var board = new Board("/dev/ttyAMA0", {
  skipCapabilities: true,
  pinCount: 66,
  analogPins: [54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65]
});

board.on("ready", function() {
  var pin = 13;
  var state = 1;

  board.pinMode(pin, board.MODES.OUTPUT);

  board.on("string", function (message) {
    console.log(message);
  });

  setInterval(function() {
    board.digitalWrite(pin, (state ^= 1));
  }, 500);
});
```

That should hopefully log:
Due ready
DW 2 : 1
DW 2 : 0
DW 2 : 1
DW 2 : 0
etc

soundanalogous, 2018-04-07:
actually it may be DW 2 : 16 rather than DW 2: 1

overflowz, 2018-04-07:
Well, to be honest it does not print anything. It just hangs. If I add DEBUG flag to see if there's something going, it's the same, transmission is happening, yet nothing is being logged without it.

soundanalogous, 2018-04-07:
But you still get:
READY
StandardFirmata.ino-2.5
?

soundanalogous, 2018-04-07:
Ah, I forgot to add that in the script.

soundanalogous, 2018-04-07:
Updated script to add that logging as well.

overflowz, 2018-04-07:
Yup. added by myself, no worries :)

[image]

It just hangs here.

soundanalogous, 2018-04-07:
in firmata.js, add `console.log("reportversion");` just after [line 738](https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L738).

Then add `console.log("skipCapabilities");` just after [this line](https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L748).

Also add `console.log(data);` [here(https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L616)].

overflowz, 2018-04-07:
output:

````
data <Buffer f9 02 05 f0 79 02 05 53 00 74 00 61 00 6e 00 64>
reportversion
data <Buffer 00 61 00 72 00 64 00 46 00 69 00 72 00 6d 00 61 00 74 00 61 00 2e 00 69 00 6e 00 6f 00 f7 f0 71 44 00 75 00 65 00 20 00 72 00 65 00 61 00 64 00 79 00 ... >
skipCapabilities
READY
StandardFirmata.ino-2.5
````

soundanalogous, 2018-04-07:
Also add `console.log("toggle");` in the setInterval block in test.js to see if the script is still running.

overflowz, 2018-04-07:
it does output toggle function every 500ms.

overflowz, 2018-04-07:
It was auto generated elipsis. I'll give you full output of that

soundanalogous, 2018-04-07:
And you have these lines in your test script?
```
board.on("string", function (message) {
   console.log(message);
});
```

overflowz, 2018-04-07:
Woops, I don't had that.

Output including on string event:
````
data f90205f07902055300740061006e0064
reportversion
data 006100720064004600690072006d006100740061002e0069006e006f00f7f071440075006500200072006500610064007900f7
skipCapabilities
READY
StandardFirmata.ino-2.5
Due ready
toggle
toggle
toggle
toggle
toggle
toggle
...
````

soundanalogous, 2018-04-07:
And do you have to hit Reset on the board to get that or does it also work without reset?

overflowz, 2018-04-07:
I have to click RESET all the time or it just hangs.

soundanalogous, 2018-04-07:
I've added another message to the logging Sketch. Copy [this gist again](https://gist.github.com/soundanalogous/e10be1c5c0e4031e5aba0769ce5bc1e8), compile and upload. That should print "ping" every 10 seconds in the JS console if the Sketch is still running on the board.

soundanalogous, 2018-04-07:
You can remove the "toggle" log as well since we know the JS script isn't hanging.

overflowz, 2018-04-07:
output when clicking RESET:
````
data f90205f07902055300740061006e0064
reportversion
data 006100720064004600690072006d006100740061002e0069006e006f00f7f071440075006500200072006500610064007900f7
skipCapabilities
READY
StandardFirmata.ino-2.5
Due ready
data f071700069006e00
data 6700f7
ping
data f071700069006e00
data 6700f7
ping
data f071700069006e00
data 6700f7
ping
data f071700069006e00
data 6700f7
ping
data f071700069006e00
data 6700f7
ping
````

output without clicking RESET:
````
data f071700069006e006700f7
data f071700069006e00
data 6700f7
data f071700069006e00
data 6700f7
data f071700069006e00
data 6700f7
data f071700069006e00
data 6700f7
data f071700069006e00
data 6700f7
````

soundanalogous, 2018-04-07:
So strange. Absolutely no issues in receiving data from the board, but data is never received by the board.

overflowz, 2018-04-07:
Can it be caused if pin is broken? For example RX of arduino, but I would not able to write sketch there too, right?

My question is, do you have the same setup and does it work for you?

soundanalogous, 2018-04-07:
I have not tried to replicate the setup yet. I've never attempted to connect an Arduino to a Pi.

overflowz, 2018-04-07:
Oh well. Got any suggestions what should I try in general? Like buy new boards, try on different ones, etc.

soundanalogous, 2018-04-07:
Well, it sounds like it works fine over USB, so I suspect there's a weird timing issue when using the UART (as evident by you needing to hit the reset button on the board). Maybe the UART isn't true full duplex or there's a weird delay in it. You could try lower baud rates like 9600 or 19200. You could also try johnny-five which is built on top of firmata.js (when an Arduino is involved). This document may be helpful: https://gist.github.com/ajfisher/8880882.

I'll let you know if I get around to setting this up locally and have any success.

soundanalogous, 2018-04-07:
If you have a different Arduino than a Due that is also worth testing. Due has been one of the more problematic boards over the years.

overflowz, 2018-04-07:
I was following that guide too, but it was written for rpi 2 though. My main project is written in johnny-five, but when I was getting firmata errors from there, thought would be great to try to start debugging from firmata.js. 

Alright, thanks! I'll also try on other boards and will let you know if I get something.

Really appreciate your help!

soundanalogous, 2018-04-07:
There are comments at the bottom for Pi 3, but it sounds like you have already tried that.

overflowz, 2018-04-07:
Yeah, I did. I tried many ways and was not able it to work tho. I'll test if lowering baud rate will help now and if not then I'm out of ideas on this and will try it on Mega later, not sure when though.

overflowz, 2018-04-07:
output of baud 9600 is even weirder:
````
data 00
data e000e000e000e000e0e00000e0e000e000e000e0e0
data e000
data 0000
data e0e00000
data e000e00000
data e0e000e000e0e000
data 00e0e000
data e000e000
data 00e0e000
data 0000e000
data e000e000
data 0000e000
data 0000e000
data 0000e000
data e000e0e000
data 0000e000
data e0e0e0e000
data e000e000
data 00e0e000
data e000e000
data 00e00000
data e000e0e000
data 00e0e000
data e0e0e000
data e0e000
data e000e00000e000
data e0e0e0e000
data e0e000e000
data 00
data 0000
data 0000e000
data e0e000e000
data e000e000
data 0000e000
data e000e000
data e0e0
````

no board ready event fired, no reaction on RESET button, nothing.

same for 115200 baud:
````
data de0e86b8f8
data de0e86b8f8
data 87f2380098a8a898
data c8a8a898c8b898e8
data 982898a8b8f8de0d
data b808c8b898a8d8f8
data de0e86b8f8
data de0e86b8f8
````

I also set option "baud" of board options in test.js as well.

overflowz, 2018-04-07:
Never mind, baud is not board's property. Tired for now, will continue testing later.

Will post if I get success anyhow.

overflowz, 2018-04-08:
Seems it worked pretty well on Arduino Mega, even no need to click RESET button. Both johnny-five, the test.js you gave and repl.js started working fine. I guess it's problem with the Arduino Due itself.

Please, if you manage to test it out on Due, post any updates here, thank you.

soundanalogous, 2018-04-08:
Which OS are you running on your Pi 3?

[12 more replies]