firmata/firmata.js #190: Make this repo a monorepo with Lerna

rwaldron, 2018-09-20:
- [ ] Publish firmata@1.0.0 in it's exact present form.
- [ ] Add lerna.json
- [ ] Move current firmata.js code to packages/*
- [ ] Split firmata.js into two packages:
  - [ ] **firmata-core**: this is most of the current contents of this project, but without an implicit `node-serialport` dependency
  - [ ] **firmata.js**: this is just a simple wrapper that "curries" `Firmata` with `Serialport`.
- [ ] Publish firmata@2.0.0 and firmata-core@2.0.0

dtex, 2018-09-20:
Two thoughts:

1. In order to better align with the defacto plugin-io ecosystem naming conventions and reduce confusion between similarly named firmata and firmata.js (which publishes as "firmata" on npm) I propose that we put the mono-repo in a new firmata-io repo and change the package name to "firmata-io". 
1. I worry that adding lerna raises the bar for new contributors but I'm willing to ride right on by this particular windmill.

rwaldron, 2018-09-20:
@dtex that would mean making a "firmata-io" repo in https://github.com/firmata/, which I think would be strange. I looked at https://github.com/node-serialport/node-serialport and there is no new repo made there.

closed 2020-06-30 by rwaldron