firmata/firmata.js #196: serialWrite does not appear to work

daftfox, 2019-04-27:
I have been breaking my head over this for quite some time am unable to figure out what the problem could be.

I'm developing a TypeScript application to control multiple firmata compatible devices over WiFi or the serial interface. One of the use-cases I have requires me to send data from a microcontroller to a secondary device connected over UART. The `serialWrite` method should be perfect for this, were it not that I don't see anything happening on the TX or RX lines. Not with an arduino uno running the default serialpassthrough sketch and not even when hooked up to an oscilloscope. The device I am using is the ESP8266 (Wemos D1).

code is as follows:

```
const serialOptions = {
    portId: firmata.SERIAL_PORT_IDs.SW_SERIAL0,
    baud: 9600,
    rxPin: 4,
    txPin: 5
};

firmata.serialConfig( serialOptions );
firmata.serialWrite( firmata.SERIAL_PORT_IDs.SW_SERIAL0, [0x01, 0x03, 0x03, 0x07] );
```

I'm running the code above after the 'ready' event has been emitted by firmatajs. The device is currently connected over WiFi.

I have flashed configurable firmata with support for serial connection to the device in question.
Would anyone be able to help me out? I'm kinda at a loss here.

soundanalogous, 2019-04-27:
Are you able to run a simple blink application to ensure the issue is with serialWrite and not with the wifi connection to the board?

soundanalogous, 2019-04-27:
Ah, [software serial is not currently supported for the ESP8266](https://github.com/firmata/arduino/blob/master/utility/SerialFirmata.h#L26). You may be able to use HW_SERIAL1 if the board is configured to enable UART1. Otherwise, it looks like ESP8266 has a software serial implementation so it may be possible to add another condition [here](https://github.com/firmata/arduino/blob/master/utility/SerialFirmata.h#L29) to include the proper software serial library for an ESP8266. If you try that and it works, please submit a pull request.

daftfox, 2019-04-27:
@soundanalogous thanks for the quick reply. Pretty much everything seems to work, except for the `serialWrite` method. Haven't tried out the `serialRead` yet, but I figured I'll burn that bridge when I get there.

I'll try out your suggestion and add an extra condition to the `serialFirmata.h` dependency. I'll keep you updated when I know more.

daftfox, 2019-04-27:
So, ever since I added the ESP8266 condition to the `SerialFirmata.h` file I can see something has changed. For starters the TX line reads ~3.3v now, so the mcu has actually setup the pins for SoftwareSerial. However I still don't see any activity on the RX line. Any more clues?

daftfox, 2019-04-27:
Hey, that did the trick! I was a little confused about not seeing anything show up in the serial monitor, but that was because I wasn't sending bytecodes corresponding with ascii characters. I will submit a pull request.

soundanalogous, 2019-04-28:
This has been fixed in the Firmata repo: https://github.com/firmata/arduino/pull/420.

closed 2019-04-28 by soundanalogous