firmata/firmata.js #209: encodeCustomFloat: "must encode custom floats when they are less than 1" is failing

rwaldron, 2019-12-28:
@dtex I don't understand this failure :|

```
1) Numeric encoding/decoding and formatting
       must encode custom floats when they are less than 1:

      AssertionError: expected Array [ 69, 2, 0, 20 ] deepEqual Array [ 79, 46, 70, 5 ]
      + expected - actual

       [
      -  69
      -  2
      -  0
      -  20
      +  79
      +  46
      +  70
      +  5
       ]

```

rwaldron, 2019-12-28:
This appears to only occur locally for me 🤷🏼‍♂️

dtex, 2019-12-28:
I'm able to replicate in node 12, but works fine in node 10. I can't imagine what has changed but I'll dig into it.

dtex, 2019-12-28:
**Tired:** Off by 1 errors
**Wired:** Off by 0.00000000000000000001 errors

Here's the scoop:

In node 10:
````js
Math.pow(10, -4) === 0.0001
````

but in node 12
````js
Math.pow(10, -4) === 0.00009999999999999999
````

Appears to be [this issue](https://bugs.chromium.org/p/v8/issues/detail?id=9622) which will be fixed by [this commit](https://chromium-review.googlesource.com/c/v8/v8/+/1774898) and will eventually make its way to node.js. If we need to, we can comment out that test for now, or we could just leave this issue open and findable until node has the corrected version of v8.

rwaldron, 2020-01-06:
Thanks for checking into that! I say we take no action on our end and leave this open for folks to easily find.