firmata/firmata.js #212: Firmata - Multiple SoftwareSerial

zerobgk, 2020-01-18:
Hey, guys,

I'm trying to communicate serial devices (pins rx / tx) on the arduino with StandardFirmataPlus, in NodeJS using the "firmata" module, do the following:

>var firmata = require("firmata");
>
>var board = new firmata('/dev/ttyUSB0');
>
>board.on("ready", function () {
>
>  console.log('Board is ready');
>  
>  var SerialPort = board.SERIAL_PORT_IDs.DEFAULT;
>
>  board.serialConfig({
>    portId: SPort,
>    baud: 115200,
>    rxPin: 50,
>    txPin: 52
>  });
>
>  board.serialRead(SPort, 1024, function(data) {
>
>    if (data) {
>      console.log(new Buffer.from(data).toString("ascii"));
>      board.serialWrite(SPort, new Buffer.from('Hello', 'ascii'));
>    }    
>
>  });
>
>});

I saw that I need to use a serial port ID, would this be for communication with the host computer?

But what I really want to know is: how can I connect more than one SoftwareSerial device like that?

zerobgk, 2020-01-18:
I'm using the serial for YS-IRTM infrared module.

dtex, 2020-01-18:
Arduino Uno?

```serialPort``` is unused and ```SPort``` is undefined. Are these supposed to be the same?

zerobgk, 2020-01-18:
Sorry I forgot to get the updated code,

here it is:

>var firmata = require("firmata");
>
>var board = new firmata('/dev/ttyUSB0');
>
>board.on("ready", function () {
>  console.log('Board is ready');
>  
>  var SerialPort = board.SERIAL_PORT_IDs.DEFAULT;
>
>  board.serialConfig({
>    portId: SerialPort,
>    baud: 115200,
>    rxPin: 50,
>    txPin: 52
>  });
>
>  board.serialRead(SerialPort, 1024, function(data) {
>    if (data) {
>      console.log(new Buffer.from(data).toString("ascii"));
>      board.serialWrite(SerialPort, new Buffer.from('Hello', 'ascii'));
>    }    
>  });
>
>  
>});

dtex, 2020-01-18:
Oh wait, you're on a Mega right?

dtex, 2020-01-18:
On a mega I would try to use one of the hardware serial ports. Software serial is not reliable at 115,200. You wouldn't want to go over 57,600.

dtex, 2020-01-18:
Here's the code updated to use hardware serial. Looking at [this](https://github.com/firmata/arduino/blob/master/Boards.h#L204-L209) I can see that hardware serial 1 is attached to pins 18(TX) and 19(RX). You don't have to specify them.

````javascript
const firmata = require("firmata");
const board = new firmata('/dev/ttyUSB0');
const SerialPort = board.SERIAL_PORT_IDs.HW_SERIAL1;

board.on("ready", () => {
  console.log('Board is ready');
  
  board.serialConfig({
    portId: SerialPort,
    baud: 115200
  });

  board.serialRead(SerialPort, 1024, (data) => {
    if (data) {
      console.log(new Buffer.from(data).toString("ascii"));
      board.serialWrite(SerialPort, new Buffer.from('Hello', 'ascii'));
    }
  });
});
````

zerobgk, 2020-01-18:
Yeah man, it works. But how i do it with others pins, just declare in serialConfig() options?

zerobgk, 2020-01-18:
Can i use another pins?

dtex, 2020-01-18:
Yes, but...

You can use any GPIO pin, but only with software serial. Software serial is limited to 57,600 baud. Can the device you are talking to be configured to use <=57,600? If not then you'll need to use one of the hardware pin combinations (14 & 15, 15 & 17, or 18 & 19).

zerobgk, 2020-01-18:
The infrared module works at 9600 baud rate. No problem, right?

zerobgk, 2020-01-18:
[image]

That's the module.

dtex, 2020-01-18:
Right, 9600 should be fine. I've used software serial @ 9600 baud with GPS breakouts no problem.

````js
const firmata = require("firmata");
const board = new firmata('/dev/ttyUSB0');
const serialPort = board.SERIAL_PORT_IDs.SW_SERIAL0;

board.on("ready", () => {
  console.log('Board is ready');
  
  board.serialConfig({
    baud: 9600,
    rxPin: 50,
    txPin: 52
  });

  board.serialRead(SerialPort, 1024, (data) => {
    if (data) {
      console.log(new Buffer.from(data).toString("ascii"));
      board.serialWrite(SerialPort, new Buffer.from('Hello', 'ascii'));
    }
  });
});
````

dtex, 2020-01-26:
According to this [spec sheet](https://docplayer.net/101050357-Nec-infrared-codec-module-ver1-0-ys-irtm.html), 9600 is the default speed.

closed 2023-01-28 by brunicius