firmata/firmata.js #217: Can I use webpack to bundle this firmata.js module into my web project?

TyutWzz-beep, 2020-03-16:
Hi, 
I'm trying to use this firmata.js module on Scratch3 web app. I wish to communicate my Arduino board directly from google chrome through the serial port. 

When I require firmata.js using const Firmata = require("firmata"); I got the following errors from cmder window after executing the webpack-dev-server --https command:
```text
ERROR in ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/linux-list.js
    Module not found: Error: Can't resolve 'child_process' in 'D:\Scratch3withFaceapi\scratch-gui\node_modules\scratch-vm\node_modules\@serialport\bindings\lib'
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/linux-list.js 1:21-45
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/linux.js
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/index.js
     @ ./node_modules/scratch-vm/node_modules/serialport/lib/index.js
     @ ./node_modules/scratch-vm/node_modules/firmata/lib/com.js
     @ ./node_modules/scratch-vm/node_modules/firmata/lib/firmata.js
     @ ./node_modules/scratch-vm/src/extensions/scratch3_arduino/index.js
     @ ./node_modules/scratch-vm/src/extension-support/extension-manager.js
     @ ./node_modules/scratch-vm/src/virtual-machine.js
     @ ./node_modules/scratch-vm/src/index.js
     @ ./src/containers/controls.jsx
     @ ./src/playground/blocks-only.jsx

    ERROR in ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/unix-read.js
    Module not found: Error: Can't resolve 'fs' in 'D:\Scratch3withFaceapi\scratch-gui\node_modules\scratch-vm\node_modules\@serialport\bindings\lib'
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/unix-read.js 1:11-24
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/darwin.js
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/index.js
     @ ./node_modules/scratch-vm/node_modules/serialport/lib/index.js
     @ ./node_modules/scratch-vm/node_modules/firmata/lib/com.js
     @ ./node_modules/scratch-vm/node_modules/firmata/lib/firmata.js
     @ ./node_modules/scratch-vm/src/extensions/scratch3_arduino/index.js
     @ ./node_modules/scratch-vm/src/extension-support/extension-manager.js
     @ ./node_modules/scratch-vm/src/virtual-machine.js
     @ ./node_modules/scratch-vm/src/index.js
     @ ./src/containers/controls.jsx
     @ ./src/playground/blocks-only.jsx

    ERROR in ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/unix-write.js
    Module not found: Error: Can't resolve 'fs' in 'D:\Scratch3withFaceapi\scratch-gui\node_modules\scratch-vm\node_modules\@serialport\bindings\lib'
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/unix-write.js 1:11-24
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/darwin.js
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/index.js
     @ ./node_modules/scratch-vm/node_modules/serialport/lib/index.js
     @ ./node_modules/scratch-vm/node_modules/firmata/lib/com.js
     @ ./node_modules/scratch-vm/node_modules/firmata/lib/firmata.js
     @ ./node_modules/scratch-vm/src/extensions/scratch3_arduino/index.js
     @ ./node_modules/scratch-vm/src/extension-support/extension-manager.js
     @ ./node_modules/scratch-vm/src/virtual-machine.js
     @ ./node_modules/scratch-vm/src/index.js
     @ ./src/containers/controls.jsx
     @ ./src/playground/blocks-only.jsx

    ERROR in ./node_modules/scratch-vm/node_modules/bindings/bindings.js
    Module not found: Error: Can't resolve 'fs' in 'D:\Scratch3withFaceapi\scratch-gui\node_modules\scratch-vm\node_modules\bindings'
     @ ./node_modules/scratch-vm/node_modules/bindings/bindings.js 5:9-22
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/win32.js
     @ ./node_modules/scratch-vm/node_modules/@serialport/bindings/lib/index.js
     @ ./node_modules/scratch-vm/node_modules/serialport/lib/index.js
     @ ./node_modules/scratch-vm/node_modules/firmata/lib/com.js
     @ ./node_modules/scratch-vm/node_modules/firmata/lib/firmata.js
     @ ./node_modules/scratch-vm/src/extensions/scratch3_arduino/index.js
     @ ./node_modules/scratch-vm/src/extension-support/extension-manager.js
     @ ./node_modules/scratch-vm/src/virtual-machine.js
     @ ./node_modules/scratch-vm/src/index.js
     @ ./src/containers/controls.jsx
     @ ./src/playground/blocks-only.jsx
```
I've done npm install -g firmata and it works fine on terminal, but I can't bundle it into my Sratch3 project using webpack.

Any suggestions?

TyutWzz-beep, 2020-03-16:
I've tried to add the 'firmata' as an external model in webpack.config.js , when I run webpack-dev-server --https under scratch-gui dictionary it then works fine, but when I run yarn run watch command under scratch-vm dictionary, errors are as the following:
```text

    ERROR in ./node_modules/@serialport/bindings/lib/linux-list.js
    Module not found: Error: Can't resolve 'child_process' in 'D:\Scratch3withFaceapi\scratch-vm\node_modules\@serialport\bindings\lib'
     @ ./node_modules/@serialport/bindings/lib/linux-list.js 1:21-45
     @ ./node_modules/@serialport/bindings/lib/linux.js
     @ ./node_modules/@serialport/bindings/lib/index.js
     @ ./node_modules/serialport/lib/index.js
     @ ./node_modules/firmata/lib/com.js
     @ ./node_modules/firmata/lib/firmata.js
     @ ./src/extensions/scratch3_arduino/index.js
     @ ./src/extension-support/extension-manager.js
     @ ./src/virtual-machine.js
     @ ./src/index.js
     @ ./src/index.js-exposed
     @ ./src/playground/benchmark.js

    ERROR in ./node_modules/@serialport/bindings/lib/unix-write.js
    Module not found: Error: Can't resolve 'fs' in 'D:\Scratch3withFaceapi\scratch-vm\node_modules\@serialport\bindings\lib'
     @ ./node_modules/@serialport/bindings/lib/unix-write.js 1:11-24
     @ ./node_modules/@serialport/bindings/lib/linux.js
     @ ./node_modules/@serialport/bindings/lib/index.js
     @ ./node_modules/serialport/lib/index.js
     @ ./node_modules/firmata/lib/com.js
     @ ./node_modules/firmata/lib/firmata.js
     @ ./src/extensions/scratch3_arduino/index.js
     @ ./src/extension-support/extension-manager.js
     @ ./src/virtual-machine.js
     @ ./src/index.js
     @ ./src/index.js-exposed
     @ ./src/playground/benchmark.js

    ERROR in ./node_modules/@serialport/bindings/lib/unix-read.js
    Module not found: Error: Can't resolve 'fs' in 'D:\Scratch3withFaceapi\scratch-vm\node_modules\@serialport\bindings\lib'
     @ ./node_modules/@serialport/bindings/lib/unix-read.js 1:11-24
     @ ./node_modules/@serialport/bindings/lib/darwin.js
     @ ./node_modules/@serialport/bindings/lib/index.js
     @ ./node_modules/serialport/lib/index.js
     @ ./node_modules/firmata/lib/com.js
     @ ./node_modules/firmata/lib/firmata.js
     @ ./src/extensions/scratch3_arduino/index.js
     @ ./src/extension-support/extension-manager.js
     @ ./src/virtual-machine.js
     @ ./src/index.js
     @ ./src/index.js-exposed
     @ ./src/playground/benchmark.js

    ERROR in ./node_modules/bindings/bindings.js
    Module not found: Error: Can't resolve 'fs' in 'D:\Scratch3withFaceapi\scratch-vm\node_modules\bindings'
     @ ./node_modules/bindings/bindings.js 5:9-22
     @ ./node_modules/@serialport/bindings/lib/win32.js
     @ ./node_modules/@serialport/bindings/lib/index.js
     @ ./node_modules/serialport/lib/index.js
     @ ./node_modules/firmata/lib/com.js
     @ ./node_modules/firmata/lib/firmata.js
     @ ./src/extensions/scratch3_arduino/index.js
     @ ./src/extension-support/extension-manager.js
     @ ./src/virtual-machine.js
     @ ./src/index.js
     @ ./src/index.js-exposed
     @ ./src/playground/benchmark.js
```
Is this because serialport dependency can't work in webpack? Is there a way to bypass this dependency?

rwaldron, 2021-08-30:
You need to provide the node polyfills https://www.npmjs.com/package/node-polyfill-webpack-plugin

closed 2021-08-30 by rwaldron