firmata/firmata.js #230: Write to Analog pins?

n00ne1mportant, 2020-12-02:
Hi
I'm using Arduino Uno board to run AlphaBot (link [here](https://www.waveshare.com/wiki/AlphaBot))
I'm trying to write to analog pins A0 and A1 but it just won't work. Ideally, I would write PWM but I can work with analog pins as digital outputs. This is supported in Arduino IDE but the pin mode must be "OUTPUT".
Reference: https://www.arduino.cc/en/Tutorial/Foundations/AnalogInputPins

I tried putting them in output mode and writing but no use. This is what I've tried:
```js
  board.pinMode(18, board.MODES.OUTPUT)
  board.digitalWrite(18, board.HIGH)
```
and this
```js
  board.pinMode(18,board.MODES.PWM)
  board.analogWrite(18, 100)
```
and this throws an error even though the analog mode should be supported (or I messed up pin numbers):
```js
  board.pinMode(18,board.MODES.ANALOG)
  // (node:2144) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'mode' of undefined 
  board.analogWrite(18, 100) // this line never runs
```
PWM works on pin 5 so this is fine:
```js
  board.pinMode(5,board.MODES.PWM)
  board.analogWrite(5,100)
```
Expected output is HIGH or LED lighting up (I hooked them all to LEDs to test it out).
Actual output is LOW ie LED is not lighting up. The one for PWM 5 does and I checked the wiring and tried switching it around.
The PWM one is the only one on HIGH. 
Is is at all possible to write to analog pins with this lib?
I got the pin numbers by running `console.log(board.analogPins)`
Thanks!

jguille2, 2020-12-02:
Hi Ivan @n00ne1mportant

I think you are confusing pwm and analog features.

`analogWrite` Arduino command (and also `analogWrite` Firmata one , that fires it) writes a 'pwm wave' to a pin. It is not a complete analog output. And it is not about 'analog input' pins.

Then, pin 18 (A4) has not "pwm" features in Arduino Uno.

See docs: https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/

Joan

n00ne1mportant, 2020-12-02:
Hello!
Thank you for responding.
I'm not confusing it but given my badly formed question I can see it looks that way :D

I just want to use analog pins for digital output. They are hardwired in the Alphabot board and I can't use other pins. 
However it doesn't work, like I said in the question. I'm starting to blame the board (hardware) at this point but I'm not sure yet. Is it possible to use analog pins as digital in this library?

Thanks again for responding

jguille2, 2020-12-02:
Hi Ivan @n00ne1mportant ,

But then, your first code
```
  board.pinMode(18, board.MODES.OUTPUT)
  board.digitalWrite(18, board.HIGH)
```
is not running for you?
This is oK!

Note that A4 and A5 are also I2C ports in Arduino UNO. Maybe they are connected to other electronic components?

Joan

n00ne1mportant, 2020-12-02:
They shouldn't be connected to anything according to the docs.
So the code I tried is correct but it won't run. 
Okay so the problem is not in the lib, I'll try to write Arduino C version to see if that runs... If not it's gotta be the board issue.

Anyways, thank you for responding!
I'll close the issue as soon as test it out.

n00ne1mportant, 2020-12-02:
I wrote .ino file and it runs fine which brought me back to thinking the problem was this lib.
So I reflashed StandardFirmata and when I tried to reproduce the bug it couldn't... so now it works.
Reflashing the Firmata must have somehow fixed it since I didn't really change anything... Just ran the same code again
Weird...

In any case thank you very much for responding!

closed 2020-12-02 by n00ne1mportant