firmata/firmata.js #234: Firmata.JS unplug detection broken

MattTavares, 2021-03-23:
The upstream serialport library does not document or seemingly support the event.disconnect as a valid error in the stream interface. The following line:
    if (event && event.disconnect && event.disconnected) {
(@ https://github.com/firmata/firmata.js/blob/54dda2d2112e9fc3f997324df22d59f6e3d05298/packages/firmata-io/lib/firmata.js#L597) never triggers because only event and event.disconnected are returned true. This breaks detection of devices when they have been unplugged. In my case this is breaking my ability to reconnect in the node-red-node-arduino downstream dependency after an arduino (or in my case a Teensy 4) is disconnected and reconnected.
Simply changing the line to read 
    if (event && event.disconnected) {
seems to fix everything

For reproducing my install
I am running on raspberry pi OS Buster 10, Node-Red 1.2.9, Serial 9.0.7, and the latest firmata v2.2.0

rwaldron, 2021-03-23:
@reconbot

reconbot, 2021-03-23:
That seems right to me https://serialport.io/docs/api-stream#close

rwaldron, 2021-03-23:
@MattTavares do you want to make a patch/pr?

rwaldron, 2021-08-30:
Merged patch

closed 2021-08-30 by rwaldron