firmata/firmata.js #271: Which ConfigurableFirmata version for accelstepper?

Green-Bug-Eyed-Monster, 2024-11-01:
Okay, I got the blink.js sketch to work on ConfigurableFirmata Arduino library version 2.10.1.

What's the story with the accelstepper routines? 

Can't seem to find a ConfigurableFirmata Arduino library version that supports these.

Client makes the request without error, yet the server does nothing with the stepper output pins, and won't fire the command complete callback.

GBEM👽

dtex, 2024-11-01:
Did you build with http://firmatabuilder.com/ ?

Green-Bug-Eyed-Monster, 2024-11-01:
Hi, dtex. Thanks.

No, I'm taking the safe bet and using the kitchen sink example file ConfigurableFirmata.js from each ConfigurableFirmata version's examples directory.

Tried everything from ConfigurableFirmata version 2.8.0 to 2.10.1, yet no result.

Difficult thing to debug, with I don't know how to tap the serial port, and no AT328P debugger.

GBEM👽

Green-Bug-Eyed-Monster, 2024-11-01:
I tried a firmatabuilder.com kitchen sink server built against 2.10.1. 

Here's the abridged debug output of running the firmata example stepper-accel.js on node:

DEBUG=* node stepper-accel.js 
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
Port: {
  manufacturer: '1a86',
  serialNumber: undefined,
  pnpId: 'usb-1a86_USB2.0-Ser_-if00-port0',
  locationId: undefined,
  vendorId: '1a86',
  productId: '7523',
  path: '/dev/ttyUSB0'
}
  serialport/stream opening path: /dev/ttyUSB0 +155ms
  serialport/binding-abstract open +0ms
Board constructed
  serialport/stream _read queueing _read for after open +2ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyUSB0 +12ms
  serialport/stream _read reading { start: 0, toRead: 256 } +0ms
  serialport/binding-abstract read +15ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
} +1ms

  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
  serialport/bindings/poller received "readable" +3ms
  serialport/bindings/unixRead Starting read +3ms
  serialport/bindings/unixRead Finished read 15 bytes +1ms
  serialport/stream binding.read finished { bytesRead: 15 } +4ms
  serialport/stream _write 3 bytes of data +1ms
  serialport/binding-abstract write 3 bytes +5ms
  serialport/bindings/unixWrite Starting write 3 bytes offset 0 bytesToWrite 3 +56ms
  serialport/stream _read reading { start: 79, toRead: 177 } +0ms
  serialport/binding-abstract read +0ms
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixWrite write returned: wrote 3 bytes +0ms
  serialport/bindings/unixWrite Finished writing 3 bytes +1ms
  serialport/stream binding.write write finished +1ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
} +1ms
...
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +4ms
  serialport/bindings/poller received "readable" +6ms
  serialport/bindings/unixRead Starting read +7ms
  serialport/bindings/unixRead Finished read 25 bytes +1ms
  serialport/stream binding.read finished { bytesRead: 25 } +8ms
board ready
  serialport/stream _write 11 bytes of data +1ms
  serialport/binding-abstract write 11 bytes +10ms
speed set
accel set
stepper enabled
command complete
  serialport/bindings/unixWrite Starting write 11 bytes offset 0 bytesToWrite 11 +11ms
  serialport/stream _read reading { start: 104, toRead: 152 } +2ms
  serialport/binding-abstract read +2ms
  serialport/bindings/unixRead Starting read +3ms
  serialport/bindings/unixWrite write returned: wrote 11 bytes +0ms
  serialport/bindings/unixWrite Finished writing 11 bytes +0ms
  serialport/stream binding.write write finished +0ms
  serialport/stream _writev 4 chunks of data +1ms
  serialport/stream _write 34 bytes of data +0ms
  serialport/binding-abstract write 34 bytes +1ms
  serialport/bindings/unixWrite Starting write 34 bytes offset 0 bytesToWrite 34 +1ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
} +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +5ms
  serialport/bindings/unixWrite write returned: wrote 34 bytes +0ms
  serialport/bindings/unixWrite Finished writing 34 bytes +0ms
  serialport/stream binding.write write finished +0ms
  serialport/bindings/poller received "readable" +4s
  serialport/bindings/unixRead Starting read +4s
  serialport/bindings/unixRead Finished read 5 bytes +0ms
  serialport/stream binding.read finished { bytesRead: 5 } +4s
  serialport/stream _read reading { start: 109, toRead: 147 } +0ms
  serialport/binding-abstract read +4s
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
} +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +3ms

Stepper motor outputs similarly invariant. Perplexing!

GBEM👽

Green-Bug-Eyed-Monster, 2024-11-02:
Got it.

I was using an AT328PB with an incorrectly configured boards.h file.

Changed to at AT328P, and all's well.

Awesome project, many thanks to developers.

GBEM👽

closed 2025-04-03 by Green-Bug-Eyed-Monster